qmk/qmk_firmware #1455: make dfu-util on Bash for Windows

roded, 2017-07-02:
Hi,
I'm trying to flash my Ergodox Infinity using the provided instructions (readme.md and gitbook) with the qmk_firmware repository's HEAD set to the commit of the 0.5.69 tag.
making the keyboard keymap works just fine, but when I attempt flashing the keyboard the `dfu-util.exe` cannot be found although it exists.
Any idea what I could be doing wrong?
Thanks

> rodedb@RODEDB:~/qmk_firmware/keyboards/ergodox/infinity$ make infinity-colemak-dfu-util
Making ergodox/infinity with keymap colemak and target dfu-util
>
> arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
>
>Size before:
   text    data     bss     dec     hex filename
      0   64492       0   64492    fbec ergodox_infinity_colemak.hex
>
>Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/ergodox_infinity_colemak.elf                                                        [OK]
Creating binary load file for Flash: .build/ergodox_infinity_colemak.bin                            [OK]
Creating load file for Flash: .build/ergodox_infinity_colemak.hex                                   [OK]
>
>Size after:
   text    data     bss     dec     hex filename
      0   64492       0   64492    fbec ergodox_infinity_colemak.hex
>
>/home/rodedb/qmk_utils/wsl_downloaded/dfu-util-0.9-win64/dfu-util.exe: 1: /home/rodedb/qmk_utils/wsl_downloaded/dfu-util-0.9-win64/dfu-util.exe: MZ�♥♦���@�♫▼�♫�: not found
/home/rodedb/qmk_utils/wsl_downloaded/dfu-util-0.9-win64/dfu-util.exe: 1: /home/rodedb/qmk_utils/wsl_downloaded/dfu-util-0.9-win64/dfu-util.exe: Syntax error: end of file unexpected (expecting ")")
tmk_core/chibios.mk:157: recipe for target 'dfu-util' failed
make[1]: *** [dfu-util] Error 2
Make finished with errors
../../../Makefile:496: recipe for target 'infinity-colemak-dfu-util' failed
make: *** [infinity-colemak-dfu-util] Error 1

fredizzimo, 2017-07-02:
Most likely you don't have the Creators Update for Windows 10.

Check these instructions for how to check and obtain that. https://docs.qmk.fm/build_environment_setup.html

roded, 2017-07-02:
Thanks for your reply.
I think I have it set up correctly:

>rodedb@RODEDB:~/qmk_firmware/keyboards/ergodox$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.2 LTS
Release:        16.04
Codename:       xenial

Is there anything else I can check?

fredizzimo, 2017-07-02:
https://support.microsoft.com/en-us/instantanswers/d4efb316-79f0-1aa1-9ef3-dcada78f3fa0/get-the-windows-10-creators-update

Press the "check your version of windows button". 

Ubuntu 16.04 also works in the Anniversary update, but it can't run windows executables.

roded, 2017-07-02:
Ah, my mistake. Thanks.
And many thanks for your work on the Infinity firmware.

closed 2017-07-02 by roded