qmk/qmk_firmware #1456: Ergodox Infinity LED issue on the slave-half

roded, 2017-07-02:
Hi,
Using the 0.5.69 tag, I'm experiencing an issue with the LED lights on the half of the keyboard which is not connected to the computer (same behavior for both the left and right halves).
The LED display works and updates just fine but the lighting is silent.
I'm using the provided colemak keymap if that makes a difference.

fredizzimo, 2017-07-03:
You are not the first one to report this https://github.com/qmk/qmk_firmware/issues/1415

But it doesn't occur on all keyboards and I don't have LEDs installed myself to test this myself, so currently I have no idea what's wrong. But I will read the code carefully to see if there's some potential issue.

roded, 2017-07-03:
Apologies, I mistyped LED instead of LCD. My issue is with the LCD backlights, not LED backlights.

fredizzimo, 2017-07-03:
Ah, I will re-open this then.

The underlying cause could still be the same, but since the behaviour is different, we should have two different issues.

fredizzimo, 2017-07-08:
Does anyone else experience this? Any additional information or help with debugging this issue is would greatly be appreciated.

mhweaver, 2017-07-10:
I've got the same issue - the appears LCD works on both halves, but the LCD backlight on the slave half doesn't. Not sure what extra info I can provide, other than verifying that more than one person has the problem, though...

fredizzimo, 2017-07-11:
Ah, after reading your edit, I now know what the issue is.

The behaviour is like that by design. The default visualizer tries to emulate the LEDs of the standard Ergodox, which has three LEDs, a red, a green and a blue one. It's a bit tricky to emulate that, with just two screens, so it works like this.

* If one "LED" is on, then display that color and brightness on the left screen.
* If two "LEDs" are on, display one color on one screen, and the other color on the other.
* If three "LEDs" are on alternate the color on the second screen.

But that's the default visualizer, which is used, if you don't implement your own. It's only there to display somewhat meaningful information, on keymaps without a custom one.

The easiest way to make a custom visualizer, is to use `simple_visualizer`, check out the default keymap for an example. You can also implement a completely custom visualizer for more interesting effects.

I will soon have time to write documentation, so stay tuned for that.

gabeplaysdrums, 2018-01-29:
@mhweaver could you possibly share a topic branch with the code changes you made to get the backlight color showing up on both halves?  That's the behavior I'd like but having trouble groking the changes needed.

drashna, 2018-03-25:
"by design" and fix linked.

Closing.

closed 2018-03-25 by drashna