qmk/qmk_firmware #1461: Ergodox EZ subdirectory

coderkun, 2017-07-03:
Several months ago the Ergodox EZ code has been moved to the subfolder `ez` of `/keyboards/ergodox/`.

Am I supposed to move my EZ keymap from `/keyboards/ergodox/keymaps/` to `/keyboards/ergodox/ez/keymaps/`? And should I now run the build inside `/keyboards/ergodox/ez/`?

Thanks for clarifying!

skullydazed, 2017-07-03:
You should leave your keymap where it is, unless there is some reason it would not work on the infinity. The `ez/keymaps` subdirectory is only for keymaps that do not work on other ergodoxes.

The recommended way to run make is from the top level `qmk_firmware` directory. You would run `make ergodox-ez-coderkun` to build a keymap named coderkun for the ergodox ez. You can also run it from the `keyboards/ergodox/` directory by running `make ez-coderkun`.

coderkun, 2017-07-03:
Thanks for clarifying, @skullydazed. But when I run `make ergodox-coderkun_neo2` from the `qmk_firmware` root directory, nothing happens (only a few output lines about avr-gcc version, copyright and warranty). Am I doing something wrong?

Additionally, how do I know if my keymap will work not only for the EZ but also for the Infinity (or possibly others)? I have no way to test this …

fredizzimo, 2017-07-03:
Could you paste the full make output here, please, so that we can have a look at it and hopefully tell you what's wrong.

Generally the keymap should be ok if it compiles on both keyboards. The Travis build system will check that when you send a pull request. Buit if you have the full build environment setup, you should also be able to compile the Infinity yourself, even without owning the keyboard.

coderkun, 2017-07-04:
I deleted the build files and the `.build` directory and now the build works fine. Sorry for the noise.

Though it says `Making ergodox/ez with keymap coderkun_neo2` and the resulting hex file is called `ergodox_ez_coderkun_neo2.hex` even though the keymap is located in the `ergodox` folder and not in the `ez` subdirectory.

Since everything is working I will close this issue.

closed 2017-07-04 by coderkun