qmk/qmk_firmware #1464: wsl_install.sh not working Syntax error

patrickhener, 2017-07-05:
I discovered a issue with the wsl_install.sh

Executing it results in following after the downloads:

wsl_install.sh: 32: wsl_install.sh: popd: not found
wsl_install.sh: 33: wsl_install.sh: Syntax error: "}" unexpected

I checked the file but cannot see the issue.

fredizzimo, 2017-07-05:
That's strange. The popd command above it works fine. Also other people have been using the script without problems. 

The first thing I would check is to see that you don't have any local changes with `git status`. 

Then you should check the line endings, as the most likely reason for it to break is the wrong type of line endings. Bash files don't work with Windows style line endings. That should be taken care of by our `.gitattributes` file, but it's possible that it's no working correctly for some reason.

patrickhener, 2017-07-05:
Doing a fresh clone of repo and then one run of wsl_isntall.sh results in the follwoing git status:

On branch master
Your branch is up-to-date with 'origin/master'.
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)

        modified:   ../lib/ugfx (modified content)

Untracked files:
  (use "git add <file>..." to include in what will be committed)

        FlipInstaller.exe
        qmk_driver_installer.exe

no changes added to commit (use "git add" and/or "git commit -a")

fredizzimo, 2017-07-08:
Sorry for the delay.

The output of Git looks almost normal. I assume you ran the command from the `util` directory, because it shows relative paths. 

The two executables that you are seeing `FlipInstaller.exe` and `qmk_driver_installer.exe` should not be in `util` but in `util/wsl_downloaded`. It's probably caused by the bug that you encounter in the install script. But in any case, you can delete the two files.

The modified `..\lib\ugfx` is probably because of line endings. In this case it seems like you are running the command in the normal Windows command prompt, but you probably updated the submodules in the WSL bash prompt. It's possible to have them both working correctly and deal with the same line endings, and I will document this soon. But anyway that's not the cause of the problem.

Because of the two extra files, it looks like pushd didn't succeed either and the script remained in the wrong folder. You probably didn't spot that error since there's a lot of output after that. 

Digging further into that, I noticed that the syntax error you are seeing is consistent with what you get if you run `popd` from `/bin/sh` instead of `/bin/bash`(or dash). It shouldn't do that, because there's clearly a `#!/bin/bash` shebang at the top of the file.

Which leads me to suspect that you ran the script explicitely through `sh` instead of bash. You should run it with the following command, from the root of qmk `util/wsl_intall.sh` or in alternatively in the the `util` directory with `./wsl_install.sh`?  If you run it with `sh wsl_install.sh` it won't work.

patrickhener, 2017-07-08:
That worked quite well. Thanks. 
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

closed 2017-07-08 by fredizzimo