qmk/qmk_firmware #1466: Error 5

iskalter, 2017-07-06:
Hi! This is my first time trying to flash a keyboard. deltasplit75 and I am getting an error trying to compile. I am a newbie at this and would really appreciate any help.

I followed the instructions at: https://docs.qmk.fm/build_environment_setup.html

Now when I try to verify my installation and run make command like what the example said, I am getting an error.

I get this twice
[image]

Then I get this one time
[image]

And this is what it says afterwards.
[image]

Here's the stackdump info
[image]

I already have the mingw-get install msys done as I was browsing earlier to check on any soilutions in here.

Thanks

fredizzimo, 2017-07-08:
Thank you for that!

It's finally starting to clear up a long running issue that we have had. See #816, #782, #709, #791, #526 and #641 and maybe others that I have missed.

The problem is that the `MHV AVR Tools` add itself to the path, and that path contains an incompatible version of `date.exe`. Our script `1-setup-path-win.bat` and `add-paths.bat` adds the `mingw` paths to the end of the path, but because the instructions instruct you to install the avr tools before running that, then the paths end up in the wrong order.

To get the correct path, I suggest that you run `make` from the `msys` shell instead of the Windows command prompt. Start it by running `C:\MinGW\msys\1.0\msys.bat`. 

If that works we could completely remove the `1-setup-path-win.bat` scripts, which does some very horrible things anyway and update the setup instructions accordingly.

BrainMcRentals, 2017-07-08:
I was having the exact same issue, I gave up trying to compile on Windows. This solved the issue, good stuff.

rimedigit, 2017-08-09:
I was having the same issue too. Now when I run the make command I have "bin/sh dfu-programmer: command not found"

I downloaded and installed circa 7 versions of the program, I have a "dfu-programmer" folder in C:Users/NAME/dfu-programmer. 

Technically I already have a hex file, I run the make with a VirtualBox Ubuntu, but now VirtualBox doesn't connect to Internet anymore and I cannot run the makefile for my own keymap.

drashna, 2018-03-25:
If you re-run the setup script, do you still have this issue?

fredizzimo, 2018-04-07:
With msys2 or WSL this should no longer happen.

closed 2018-04-07 by fredizzimo