qmk/qmk_firmware #1472: Feedback when using DFU

AndreasBackx, 2017-07-07:
I was using my KBD75 and tried to use dfu to automatically flash it. At first I thought that it wasn't working because I wasn't getting any feedback when putting the device into reset mode. Then after a couple of seconds it flashed successfully. Could the make command have some extra feedback like:

```Found device...
Preparing to flash device...
Flashing device...
Flashed device successfully.
```

Of course the feedback would be a bit more tailored to what actually happens.

I tried to enable verbose mode, but that doesn't give any more information unfortunately.

skullydazed, 2017-07-07:
You should have some feedback, you just may not know that's what it means. If you look at the output that was generated you should have something like this at the end:

```
Bootloader Version: 0x00 (0)
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Checking memory from 0x0 to 0x47FF...  Empty.
0%                            100%  Programming 0x4800 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
Validating...  Success
0x4800 bytes written into 0x7000 bytes memory (64.29%).
```

Those messages are printed by the flashing process.

AndreasBackx, 2017-07-07:
My apologies, I should've added some more information. This is the information I get currently when issuing `make keymapname dfu`:

```shell
# Removed some compilation logs
# ...
# This first line is shown immediately:
Making kbd75 with keymap keymapname and target dfu                                                [WARNINGS]
# All of the following lines are shown immediately when the flashing was successful.
# These are not printed to me as it seems they should.
dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
Bootloader Version: 0x00 (0)
Erasing flash...  Success
Checking memory from 0x0 to 0x6FFF...  Empty.
Checking memory from 0x0 to 0x53FF...  Empty.
/bin/sh: line 0: printf: `%': invalid format character
Making kbd75 with keymap keymapname and target dfu
```

The text seems to indicate that it wants to show me this data immediately:
```
dfu-programmer: no device present.
Error: Bootloader not found. Trying again in 5s.
```

However this text is not shown to me til the process has finished.

There also seems to be a problem with the `printf` command being issued:
```
/bin/sh: line 0: printf: `%': invalid format character
```

When it should probably print what you posted:
```
0%                            100%  Programming 0x4800 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
0%                            100%  Reading 0x7000 bytes...
[>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
```
I ran this command in zsh and bash with the same results. I'm running Arch Linux 4.11.6-3 with zsh 5.3.1.

Maybe there is a problem then with how it works on my machine?

fredizzimo, 2017-07-08:
That seems strange, I don't see any `printf` calls at all in that particular part of the makefiles. So it could come from the `dfu-programmer` itself. You could try running `dfu-programmer` manually. 
```
dfu-programmer atmega32u4 erase --force
dfu-programmer atmega32u4 flash  .build/kbd75_keymapname.hex
```
Run these commands from the root of qmk, or alternatively use a relative path to the `.build` folder.

It could also be because you have the syntax of your makefile command slightly wrong, it should be `make keymapname-dfu` and not `make keymapname dfu`, I think the second variation would first build the keympaname and then flash it, followed by building the default keymap and flashing it, the second version just builds and flashes the keymapname.

Wraul, 2017-07-09:
I can confirm that I get the same behaviour as @AndreasBackx if I run the following `make <keymap> dfu` but it works correctly if I run `make dfu KEYMAP=<keymap>` or `make <keymap>-dfu`.

Is this documented somewhere? I wasn't able to find it anywhere. Though this doesn't mean that it doesn't exist somewhere.

closed 2017-10-22 by skullydazed