qmk/qmk_firmware #1474: KC60 Missing Details on Wiring

axtran12, 2017-07-08:
I know that this isn't really code (and I'm sorry for it) but it looks like a previous check-in may have wiped out the KC60 config details on WS2812B wiring? Could someone correct that if they have the details?

It would probably be also be good to include where everything is in text rather than just via picture.

fredizzimo, 2017-07-08:
Yes, for the main keyboard readme doesn't seem to be updated with a link to an external image, like it was for other images.

However, there's a special keymap with all the information, check this out 
https://github.com/qmk/qmk_firmware/tree/master/keyboards/kc60/keymaps/ws2812

axtran12, 2017-07-08:
Ahh, thanks. Looks like it might be beneficial to link from the main page to the ws2812 page :)

Also, is this an appropriate place to discuss the hardware layout of that PCB? e.g. 2nd left on bottom row LED silkscreen for positive terminal is incorrect, what does PF5 do other than exist at that tap point, which/where's Caps Lock's pin, etc.

fredizzimo, 2017-07-08:
I don't think this is the appropriate place to discuss the hardware at least if you want to reach the creators of it. It seems like NPKC, created it, but I'm not sure if they have any other support than the massdrop discussion. So I suggest that you ask the questions there.

You can also try asking on the [QMK Gitter](https://gitter.im/qmk/qmk_firmware), maybe someone knows better there.

axtran12, 2017-07-08:
I wasn't really thinking discussion, but rather notes (I guess we could add comments to config files) regarding a few things to keep in mind when configuring QMK.

skullydazed, 2017-07-08:
If we don't have a good place for information like that in the current readme format it strikes me as something we probably want to carve out a new place for. Maybe the readme needs a "Hardware Notes" section, which could describe this (and things like the differences between rev1/2, or ez and infinity.)

drashna, 2018-10-21:
This specific issue seems to be addressed.

However, the hardware into ...

closed 2018-10-21 by drashna