qmk/qmk_firmware #1476: QMK processes only one key each scan loop

fredizzimo, 2017-07-08:
QMK processes only one key each scan loop, which could cause keys to be reported in the wrong order in certain cases. 

If you press two or more keys at the same time, then the first key in the order defined by the `KEYMAP` is always processed first and sent to the computer first. The next key is sent the next scan_loop in the different USB report.

This can cause problems, for example with modifiers, if you press the modifier and the key at almost the same time, then the key could be sent before the modifier, which gives the wrong result. But also for example midi chords, plover support, or games, which requires simultaneous keypresses might cause problems. For example, for midi this would cause the tones of a chord to not start at the same time.

The effect is magnified if the keyboard has a slow scan loop, or a high debouncing setting. Since then it becomes more likely that you press keys at the same time.

I don't believe this is a huge issue, and most cases where things get wrong are most likely assumed to be typos. 

The real fix for this would be to delay the keyboard report until all keys have been pressed. But complications might come from custom functionality that can send explict reports. I'm not sure if that use case need to be fully supported though. 

Another possible fix is to somehow be able to process multiple keys at once, but that has big implications on the code base, an an attempt to fix that would not be easy.

A third solution, which handles only the modifier case, would be to make sure that modifiers are processed before normal keys on press, and after keys for releases. That would at least make sure that the order is what the user most likely inteneded.

seebs, 2017-11-18:
So, it turns out if you just count keys processed and cap it at, say, five, instead of one, this gets a lot better. And each individual process_record is still handling only one key event.

Six might be a good target because 6KRO is common. And I'm looking from a steno perspective, where I'm actually concerned about performance for "release 15 keys at once", but even then, the factor of six speedup moves us from "slow enough to be annoying" to "imperceptible".

I'll submit a patch when I get a chance.

drashna, 2018-03-25:
Seebs fixed in #2022. Closing

closed 2018-03-25 by drashna