qmk/qmk_firmware #1477: The macro interval is double of what it should be

fredizzimo, 2017-07-08:
The macro interval, `I(ms)` which is supposed to change the stroke interval of the keypresses is applied both for keydown and keyup, which makes it twice as long as it should be. This could be a documentation issue, or it could be a real issue, but either the code or documentation should be fixed.

Another more minor issue is that the delay is also applied to the `I(ms)` statement itself, so if you start a macro with that, there will be a delay before the first key is sent.

drashna, 2018-03-25:
Macros are depreciated now.  
Should this be closed?

fredizzimo, 2018-04-07:
I'm not sure, the newer `SEND_STRING` doesn't allow you to change these intervals in the middle so that's still something that the old macros do better. So I would leave this open at least until that is implemented.

jackhumbert, 2020-03-13:
I think this can be closed - we can have a new issue for implementing intervals in send_string, but my suspision is that people will be using custom keycodes for this sort of thing.

closed 2020-03-13 by jackhumbert