qmk/qmk_firmware #1478: The tapping functionality has some really strange timing issues

fredizzimo, 2017-07-08:
Define a key for example `SFT_T(KC_P)`, the following happens. Note that you are probably not able to reproduce the sequence, since the timing is very strict. But the unit tests in `test_tapping.cpp` reproduces this 100%.

1. press the key
    * Nothing happens as it should
2. release the key, on the next scan loop
    * `p` is sent, this is epxected since tap keys have to send on release and not on press
3. press the key again
    * `p` is sent - notice how this is different from the first press
4. release the key on the next scan loop
5. wait for `TAPPING_TERM` plus 1 ms
6. press the key
    * Works as expected again, nothing is sent on press
7. release the key
    * Works as expected, `p` is sent
8. wait for `TAPPING_TERM` plus 1 ms
9. press the key
    * Shift is sent, immediatly on press!

drashna, 2018-03-25:
Is this fixed?

fredizzimo, 2018-04-07:
Yes, I'm closing this. The issue can be fixed by defining `TAPPING_FORCE_HOLD`, for some reason that's not the default. 

Without `TAPPING_FORCE_HOLD`, the second press of the same key will always be a tap, no matter how long you hold it.

closed 2018-04-07 by fredizzimo