qmk/qmk_firmware #1480: Line ending issues with recent pull

That-Canadian, 2017-07-08:
So I'm having issue with the recent QMK master update, where now git is yelling that almost all files in the lib/lufo directory are modified, even on a hard reset and checkout. It looks like this is an issue with un-normalized line endings being commited to the branch. Can anyone else confirm? I'm on Bash for Windows, here's a screencap of part of the issue : 
[image]

fredizzimo, 2017-07-08:
Yes. it's a known issue. Yesterday we were working on fixing it correctly in the LUFA repository https://github.com/qmk/lufa, but got interrupted in the process, so it never got fixed.

Sorry for the invconvenience, we will try to fix it soon.

That-Canadian, 2017-07-08:
Cool glad I'm not going crazy, will just hold back on any pull requests for now

fredizzimo, 2017-07-08:
It should be fixed now.

closed 2017-07-08 by fredizzimo