qmk/qmk_firmware #1481: Trouble accessing header files

BrainMcRentals, 2017-07-08:
I can't access any headers stored outside qmk_firmware-master. When i compile my keymap.c i get a file not found error unless i copy the header to the quantum folder. I can't figure out how to get keymap.c to include my usual header directory.

fredizzimo, 2017-07-08:
Which header do you need, because even if you get it to use other headers, you would still not be able to send pull request here? The reason is that we make sure that all keymaps compile with the default QMK installation.

But if the headers that you want are useful enough, then maybe they could be included in QMK itself, at least if they have compatible licenses.

BrainMcRentals, 2017-07-08:
I was going to use shellapi.h to launch folders and programs. Is there another way to do this? My C knowledge is pretty basic and it seemed this was the easiest way to go about it.

fredizzimo, 2017-07-08:
You won't be able to do that directly from the keyboard. We do have plans for an API, which allows it, but I don't know when that will happen. 

Meanwhile I suggest that you make the keyboard send sequences that [autohotkey](https://autohotkey.com) can intercept.

BrainMcRentals, 2017-07-08:
Ok, thanks for the info.

closed 2017-07-08 by fredizzimo