qmk/qmk_firmware #1482: Mod Tap with one shot modifiert?

jansorg, 2017-07-08:
Thank you for this wonderful project!

I tried to combine a mod tap with a one shot modifier, like `MT(MOD_LALT, OSM(MOD_RALT))`, this didn't work on my Ergodox EZ.

I'd like to get a key which send left-alt when I hold it and a one shot right-alt when I just press it once. This would be wonderful for the ergodox with a US INTL layout (right alt + a prints a "a umlaut", for example).

If there is a better place to ask this then I'm sorry for posting here.

fredizzimo, 2017-07-08:
Yes this is the right place.

Unfortunately, it's not possible to combine those modifiers at the moment. I can see how it would be useful though, so that's probably something we should work on supporting in the future.

I also don't think it's possible to activate those programmatically, otherwise you could probable implement that with `process_record_user`, see [this](https://docs.qmk.fm/custom_quantum_functions.html). It's still probably possible to do it that way, but you have to do everything manually.

jansorg, 2017-07-10:
@fredizzimo Thanks! I'm probably not able to write this from scratch, maybe I'll try to use `process_record_user` at some time.
I'm a very happy user of qmk, finally I have (almost) all the shortcuts I want in hardware :)

danielo515, 2017-12-17:
Hello @fredizzimo

> Unfortunately, it's not possible to combine those modifiers at the moment.

It's not possible because he is using left and right modifiers or because it's not possible to use MT along with OSM? 
I'm currently interested in combining MT with OSL

drashna, 2018-03-25:
Not possible because the macro can't handle the special codes, IIRC. 

Also, closing.  If needed, reopen or open new issue.

closed 2018-03-25 by drashna