qmk/qmk_firmware #1490: Different Layouts Unclear from Documentation

Wheels2050, 2017-07-11:
It is unclear in the documentation (at least, to me) how one implements different key layouts for a given keyboard. For example, the KBD75 PCB supports several different bottom row layouts, in which the total number of keys changes.

How does one know how to implement this in the KEYMAP macro? For example, one of the possible KBD layouts has 5 keys to the right of the spacebar, while another one has 6 (see https://i.redd.it/0yag0r7vk0yy.jpg vs https://i.redd.it/9oxfz1vpxbwy.jpg). Do these use different entries in the KEYMAP macro and, if so, how do you know which ones to use?

sparkyman215, 2017-08-04:
I'm not an expert on this so correct me if I'm wrong:  
You would have to change the .h file in the base folder to match the layout that you have, and then in the keymap .c files you make the specific keycodes.  
Read about it [here](https://docs.qmk.fm/understanding_qmk.html#matrix-to-physical-layout-map)

Wheels2050, 2017-08-13:
You're correct on how to edit the keymap, but my issue is that I don't think it's clear how to know which entries in the KEYMAP macro are relevant.

For example, the Enter key (on a standard layout) is 2.25 units wide, so it covers two of the positions in the layout matrix. To account for that, the KEYMAP macro uses the KC_NO value to indicate those positions where a switch is not present. So, for the enter key, one slot will be assigned KC_NO and the other will be assigned a valid keycode.

For a given keyboard, how does one know which of the matrix entries should be marked as KC_NO? Naively, it's not clear which of the two matrix entries corresponding to the positions that the Enter key covers should be used, and which shouldn't. My guess is that it has something to do with the wiring layout of the keyboard, but I haven't seen any explanation in the documentation.

Note that this isn't really an issue for someone just wanting to alter their keycodes, but is more for someone trying to add QMK support for a new keyboard.

skullydazed, 2017-08-13:
You raise good points, and documentation is something we're actively working on. I wrote this a few weeks ago in response to a ticket about the `KEYMAP()` macro, possibly this one. It addresses some of your questions:

https://docs.qmk.fm/understanding_qmk.html

We still have work to do to make this documentation easier to discover, obviously.

drashna, 2018-03-25:
Is this still an issue (eg, is the documentation change not adequate)?

Wheels2050, 2018-04-21:
I still think the documentation is unclear for this specific question. For instance, on the page:

https://docs.qmk.fm/for-a-deeper-understanding/understanding-qmk

in the matrix to physical layout section, the KC_NO is put in the macro to indicate that no switch is present. If I'm trying to implement a new PCB that doesn't already have a layout in QMK, how do I know which entries in the macro should be disabled? (In the example there, how do I know which of the entries in the matrix for the numpad are inactive?).

Is this just trial and error, a matter of consulting the PCB traces to work out which switches go where, or some other method?

drashna, 2018-04-21:
Yeah, that's a bit more advanced, I think. 
But basically, it's any point in the matrix that doesn't actually have a corresponding switch hooked up.

jackhumbert, 2020-03-13:
It sounds like the questions you had were addressed - feel free to reopen this or a PR on the docs if you think something could be clarified there. The doc linked above is now at this location: https://docs.qmk.fm/#/understanding_qmk

closed 2020-03-13 by jackhumbert