qmk/qmk_firmware #1495: Shift to custom key

Yatreyu, 2017-07-13:
This is really similar to #108 but I'm still having issues.
I used http://kbfirmware.com/ to make my ".hex".
In my situation I want to shift to "!" from the base layer "?"
I'm using OSM(MOD_LSFT) as my only shift key so I don't know how to put the pieces together here.
Thanks in advance.

fredizzimo, 2017-07-14:
I'm working on a making a general solution for this, and I will probably have something ready in a few days. 

The idea is that you will be able to define a complete key with all different modifier levels, none, shift, alt gr, and alt gr+shift. This includes the support for sending for example unshifted keys, even if you have the shift key pressed.

I have a very basic version of this in my personal keymap, but I want to make something thing that is generic and included in QMK.

Yatreyu, 2017-07-16:
I'll watch out for it. I think I've always used "+" more than "=" so being able to switch those would be really great, same with "?" being much more common than "/".

Following the philosophy of Dvorak I'm really excited to switch the "?" to the unshift layer and then switch out "!" and "/". 

I imagine several people are pretty keen on this. Thanks for working on this!!!

fredizzimo, 2017-07-17:
I was thinking of implementing the following API

* `void send_keydown(uint16_t  kc)`- Alias for `register_code16`
* `void send_keyup(uint16_t kc)` - Alias for `unregister_code16`
* `void send_keypress(uint16_t kc,  ...)` -  Sends key down, followed by key up. You can add extra `KC_ `modifiers passed as varags, for example `KC_LSFT`. Note that only one modifier per argument is allowed. These modifiers are always sent verbatim, meaning that if you for example have specified `KC_LSFT` as a verbatim modifier, shift is already pressed and want to send and unshifted key, the shift will temporarily be released during the key press. 
* `void send_modified_keypress(uint16_t unshifted_kc, uint16_t shifted_kc, uint16_t altgr_kc, uint16_t altgr_shifted_kc)` - Sends the keycode depending on which modifiers are held down, for example `unshifted_kc` if neither `shift` nor `alt gr` is pressed and `altgr_shifted_kc` when both `shift` and `alt gr` are pressed. `KC_NO` can be used to not send anything for that particular combination. The function internally calls `send_keypress(kc, KC_LSFT, KC_RSFT, KC_RALT)`temporarily be released during the key press.

This means that you can define a custom keycode in your keymap and then
```c
enum my_keycodes {
   PLUS_EQL=SAFE_RANGE
};

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    switch(keycode) {
    case PLUS_EQL:
        send_modified_keypress(LSFT(KC_EQL), KC_EQL, KC_NO, KC_NO);
        return true;
        break;
    }
    return false;
}
```

tamlyn, 2018-01-28:
Did anything come of this? If not, was it due to technical issues or lack of time?

drashna, 2018-03-25:
@tamlyn  I believe that fredizzimo didn't add this yet.

However, you can see his keymap here: 
https://github.com/fredizzimo/qmk_firmware/blob/18d83e5c98dd3a9dbf5a5e0f1579e86bee2d924f/layouts/community/ergodox/fredizzimo/keymap.c

Tylaryn, 2018-04-12:
@tamlyn @Yatreyu @fredizzimo
I've been working on implementing the desired feature, that is to say being able to customize the modified values of given keys when _Left Shift_ and/or _Right Shift_ and/or _Right Alt_ are pressed (I call them Character modifiers or **charmods** and I call the feature **Custom Modified Values** or **CMV**).
Although, I haven't searched enough before trying to code this feature and haven't noticed this Issue which could have lead me to a better approach because I'm afraid mine is pretty _messy_ and "_unclean_". To be fair to myself, it is my first ever contribution to an open-source project.
Here's the link to my fork where you can find my current implementation of the feature in `keyboards/ergodox_ez/keymaps/CMV_adya/custom_modified_values.h`: [https://github.com/Tylaryn/qmk_firmware](url)

# My approach
I **override the function** `uint16_t keymap_key_to_keycode(uint8_t layer, keypos_t key)` which is used internally by the firmware each time it needs to know **what keycode to process**. In this function, we call the **user-defined function** which returns the custom modified values for the given key, if there is any (see **Feature 3** below) and, depending on the modifiers state (see **Problem 2** below), we determine the **KCID** (for **KeyCode ID**) for the key, which is an integer from 0 to 3, **0** for the **default** value, **1** for the **shifted** value, **2** for the **altgr'd** value and **3** for the **shift+altgr'd** value. The **KCID** is used as an **index** in the keycode arrays containing the custom modified values. 
Since the `keymap_key_to_keycode` function is called several times for one key event processing, in order to avoid executing the same thing multiple times, I use the `typedef`'d structure `cached_key_t` to **cache the keycode** processed after executing `keymap_key_to_keycode` for the first time on this key event (which is also used to store the boolean for the **Transparency resolution** (see **Problem 3** below)).
Since we need to know if the key event we are dealing with is a **press or release** in order to execute the correct instructions and the **record isn't passed** to the function `keymap_key_to_keycode` as it is with `process_record_user`, I added into `tmk_core/common/action.c` a variable called `cur_record` updated every time `action_exec` is executed (i.e. for each new key event), and the function `get_current_record` which returns the variable `cur_record`.
Because of what I wanted CMV to let us do, pressing charmods while holding a key `K` must make the firmware process the release event of the keycode bound to `K` and then the press event of the custom modified keycode of `K` (See **Feature 4 - scenarii 2, 3 and 4** below). This is why I added a `key_queue` in `tmk_core/common/keyboard.c` which contains the **additional key events to process** after the current key event. To **add key events to the queue**, use `bool push_key_to_q(keyevent_describer_t new_key)` or the abstracting function `void add_to_q_release_and_press(heldkey_t* active_held_key, uint8_t new_kcid)`.
When releasing a non-modifier key, we want to return to the firmware the **same keycode as** we returned for **its press event**. In order to do that, I use an array of `heldkey_t` which is a structure that store the **key position** (`keypos_t`), the **current KCID** and the **array of keycodes** assigned to the key. So, on a releasing event, we search into that array of `heldkey_t` the key at the given `keypos_t` and we return _`array_of_keycodes[KCID]`_ to the firmware.

Now, I'd like to list the things I wanted the feature to let us be able to do as well as the problems I encountered, so my implementation can be improved or redesigned by other contributors:

# Features that CMV enables
1. The user must be able to **customize the keycodes** processed by the firmware depending on the **charmods state**. (i.e. each key can be assigned a _default_ value, a _shifted_ value, a _altgr’d_ value and a _shifted+altgr’d_ value)
2. We want the **default value (KCID=0)** of the key to be processed if any modifiers other than the charmods are pressed even if a charmod is active. This is to allow the user to still use shortcuts like `Ctrl+Shift+;` in programs while a custom shifted value has been assigned to the `;` key.
3. The user **doesn't have** to assign custom modified values to **every single key** because it is more likely that they want only a few custom ones. (_I think passing the `keypos_t` object and the current layer to the function used by the user to customize the modified values is a good approach. Here's the prototype of the function I used:_ `bool keycodes_for_key(uint16_t default_kc, uint8_t layer, uint8_t row, uint8_t col)`_. Alternatively, we could use custom keycodes combined with_ `SAFE_RANGE`)
4. There are **4** different notable scenarii to handle (_KR = Keyboard Report, AKQ = Additional Keys Queue_) (The first two are more relevant and more likely to happen):
    - A modifier key `M` is pressed down **(1)**, then a non-modifier key `K` **(2)**. `K` is then released first **(3)**, followed by `M` **(4)**:
        - **(1)** - Send `M` as a modifier in the KR.
        - **(2)** - If the key `K` has a custom `M`'d value, remove `M` from the KR and add it the custom value. Otherwise, just add the default value of `K`.
        - **(3)** - Remove the value of `K` sent during **(2)** from the KR and add it back the modifier `M` if it was removed.
[... cut ...]
I am aware that my approach and my implementation are not _great_. This is why I am posting this so contributors can help improve my existing code or create their own version of the feature (this is why I detailed all the features I wanted CMV to have and some of the problems i faced) with maybe a better approach.
Also, I didn't find any documentation on how to add a feature to the firmware so it can be enabled in the `rules.mk` files by the user. Could it be explained to me ?

I'd really like to read about what you think about my implementation as well as debating about the features CMV should enable, etc.

drashna, 2018-10-21:
Since this is a PR, I'm going to close this.

closed 2018-10-21 by drashna