qmk/qmk_firmware #1496: quantum/visualizer/default_animations.c build errors

packman86, 2017-07-14:
I am trying to build a new firmware for my Ergodox Infinity and the build doesn't complete.

Below is the message I get when I build.
__
Compiling: quantum/visualizer/default_animations.c                                                 quantum/visualizer/default_animations.c: In function 'keyframe_enable':
quantum/visualizer/default_animations.c:42:5: error: implicit declaration of function 'lcd_backlight_keyframe_enable' [-Werror=implicit-function-declaration]
     lcd_backlight_keyframe_enable(animation, state);
     ^
quantum/visualizer/default_animations.c: In function 'keyframe_disable':
quantum/visualizer/default_animations.c:55:5: error: implicit declaration of function 'lcd_backlight_keyframe_disable' [-Werror=implicit-function-declaration]
     lcd_backlight_keyframe_disable(animation, state);
     ^
quantum/visualizer/default_animations.c: In function 'keyframe_fade_in':
quantum/visualizer/default_animations.c:66:5: error: implicit declaration of function 'lcd_backlight_keyframe_animate_color' [-Werror=implicit-function-declaration]
     ret |= lcd_backlight_keyframe_animate_color(animation, state);
     ^
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
tmk_core/rules.mk:358: recipe for target '.build/obj_ergodox_infinity_doxxy/quantum/visualizer/default_animations.o' failed
make[1]: *** [.build/obj_ergodox_infinity_doxxy/quantum/visualizer/default_animations.o] Error 1
Make finished with errors
../../Makefile:498: recipe for target 'ergodox-infinity-doxxy' failed
make: *** [ergodox-infinity-doxxy] Error 1

__

This is my make command:
make ergodox-infinity-doxxy VISUALIZER_ENABLE=yes LCD_BACKLIGHT_ENABLE=yes LCD_ENABLE=yes BACKLIGHT_ENABLE=yes MASTER=left

I am sure there is a simple fix for this, I just don't have the knowhow.

Thanks,

fredizzimo, 2017-07-14:
It seems like you have ran into some merge conflicts, those functions were renamed for example from, `backlight_keyframe_animate_color` to `lcd_backlight_keyframe_animate_color`. 

And they should be renamed in this file too https://github.com/qmk/qmk_firmware/blob/master/quantum/visualizer/lcd_backlight.h, and also in the corresponding `C` file.

packman86, 2017-07-14:
That was the issue!

Thank you!!!

closed 2017-07-14 by packman86