qmk/qmk_firmware #1497: Need help: disabling a key (DEL) if another key is pressed (BSPC)

JeanOlivier, 2017-07-14:
For some reason, I want to be able to hold BSPC and keep on backspacing even if I accidentally hit del.
To do this, I've decided to simply use `process_record_kb`, as I did something similar with it before.

Here's my code, that doesn't work as is:

``` C
bool process_record_kb(uint16_t keycode, keyrecord_t *record) {
  switch (keycode){
    // Non-pertinent cases removed
    case KC_DEL:
      if (record->event.pressed && (keyboard_report->keys[KC_A])) {
        return false;
      } else {
        return true;
      }
    default: return true;   // Let QMK handle the rest as usual
  }
}
```

I suspect my error is in the `keyboard_report->keys[KC_A]` part, but I can't figure out how to use it properly.

If I change the `if` line to check for a modifier, it seems to work (LSFT+DEL → NOP): 
```
if (record->event.pressed && (keyboard_report->mods & MOD_BIT(KC_LSFT))) 
```

Any help would be greatly appreciated!

fredizzimo, 2017-07-14:
I can't really imagine situations where you hit backspace instead of del, but anyway...

First of all, you should use `process_record_user` in your keymap and not `process_record_kb`, which is common for the keyboard.

Accessing the keyboard report is a bit hackish way of doing this, as it's not really meant to be read. That said it is possible, but the way of doing it depends on if you have `NKRO`,  `USB_6KRO` or neither of them enabled. I won't explain the `NKRO` or the `USB_6KRO` cases since they are much more complicated. But you can study the `tmk_core/report.c` file to figure it out.

For the normal case, you need to check the keys in a loop, like this
```c
for(size_t i=0; i<KEYBOARD_REPORT_KEYS; i++) {
    if (keyboard_report->keys[i]==KC_BSPC) {
        return false;
   }
}
```

The other way of doing it, is to handle the key press and key up events for `KC_BSPC`, and set a boolean flag that you are holding it, then check that flag for the `KC_DEL` event. 

Finally, you probably want to block the released event as well.

JeanOlivier, 2017-07-15:
Thank you very much!

> I can't really imagine situations where you hit backspace instead of del, but anyway...
Split backspace key on 60%... Still not used to it... Don't judge me...

I have NKRO enabled, so I implemented your other suggestion. Here it is:
``` C
bool del_blocked = false; // This doesn't feel clean...

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode){
    // Cases omitted
    // If BSPC is held, DEL is disabled.
    case KC_BSPC:
      if (record->event.pressed) {
        del_blocked = true;
        return true;
      } else {
        del_blocked = false;
      }
    case KC_DEL:
      if (del_blocked) {
        return false;
      } else {
        return true;
      }
    default: return true;   // Let QMK handle the rest as usual
  }
}
```

Is there a better practice than a to define a bool in the body of the keymap file? It doesn't feel clean to me, but I don't know how else to do it. I'm not a programmer, so the subtleties of C escape me.

Also, I'm not sure if my implementation takes care of the key-up events like you mentioned?

Thank you very much again!

fredizzimo, 2017-07-16:
Have you considered switching the location of backspace and del? You most likely have backspace on the far side, which means that you will have to do quite a bit of finger streching, so when you press it many times, the finger will automatically get closer and hit the del key. On the other hand, it could be harder to hit if it's not the outmost key, so my suggestion might not work.

Defining a bool in the keymap file is perfectly fine for this, but you should declare it static, like this.
```c
static bool del_blocked = false;
```
The reason for this is that non-static symbols get external linkage, which means that you will get an error, if some other `C` file also defines the same variable.

Your code is almost correct, but `KC_BSPC` should return true in both cases, since otherwise you will miss the key up event. So it should like this
```C
    case KC_BSPC:
      if (record->event.pressed) {
        del_blocked = true;
        return true;
      } else {
        del_blocked = false;
        return true;
      }
```

But the bad news is that the code is buggy, if you do the following

1. Press del
2. Press backspace
3. Release del
4. Release backspace

In that case, I think the del key will be stuck down. So when thinking about it a bit more, I think you should be fine by not blocking the key up events, so the final version should look like this.

```c
static bool del_blocked = false; // This doesn't feel clean...

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode){
    // Cases omitted
    // If BSPC is held, DEL is disabled.
    case KC_BSPC:
      del_blocked = record->event.pressed;
      return true;
    case KC_DEL:
      if (del_blocked && record->event.pressed) {
        return false;
      } else {
        return true;
      }
    default: return true;   // Let QMK handle the rest as usual
  }
}

JeanOlivier, 2017-07-16:
Thank you very much again for your answer. 

My top-left row is MINS, EQL, BSPC, DEL. I like it, but I'm still not perfectly used to hit BSPC bullseye when typing fast at my fastest. DEL isn't a problem, being on the edge of the keyboard. 

I managed to get BSPC stuck on my previous implementation, and so far yours seems to be working perfectly. I'll mark the issue as solved.

Thanks!

closed 2017-07-16 by JeanOlivier