qmk/qmk_firmware #1499: Ergodox ez steno layout mouse doesnt work

johnwalker, 2017-07-16:
Hi, I flashed my ez with the ergodox ez steno layout. But it looks like none of the mouse buttons and navigation works on layout 2.

Maartenwut, 2017-07-16:
Change MOUSEKEY_ENABLE = no to MOUSEKEY_ENABLE = yes in your Makefile.

Talljoe, 2017-07-27:
The virtual serial port and the mousekeys feature use the same interrupt and therefore can't be enabled at the same time. It's something I'm looking into to better support the stenography change that I submitted today.

drashna, 2018-03-25:
Was this resolved?

johnwalker, 2018-05-03:
no

razajafri, 2018-06-21:
Has there been any resolution to this? 

My symbol layer isn't working either. whats the work around to that?

drashna, 2018-10-21:
It may be worth applying this PR, and see if it fixes the issues:
https://github.com/qmk/qmk_firmware/pull/3951

jackhumbert, 2020-03-13:
The endpoint limit isn't something we can really address here - if you have any code snippets or additional complications, feel free to post them here, but I think we can close this for now.

closed 2020-03-13 by jackhumbert