qmk/qmk_firmware #1500: RGB Underglow

Dbroqua, 2017-07-16:
Hello,

I'm currently working on a new custom keyboard (https://github.com/Dbroqua/DK60TP) and I'm a little bit confused about the RGB underglow.

On my PCB I have 8 leds but if in my config.h I set `#define RGBLED_NUM 8` only the 6 first leds work.

I have made some tests ( `#define RGBLED_NUM 1` to ``#define RGBLED_NUM 10`) and I have seen that the leds 3 and 7 generate the gap.

I think it's an electronic issue but why and how to fix them?

Regards,

Damien.

drashna, 2018-03-25:
I think I remember seeing this in the gitter (or maybe reddit). 

Probably a bad strip.  And if so, replacing would be the only real solution. 

Also, did you get this resolved?

Dbroqua, 2018-03-25:
Hello,

I don't use strip but a custom PCB with leds soldered one by one on the PCB.

drashna, 2018-10-21:
Sorry for the super long delay. :(

Most likely, it's a burned out LED, as the individual LEDs are very sensitive.

closed 2019-11-21 by drashna