qmk/qmk_firmware #1501: keyboard.c is ignored in keymap folder

Maartenwut, 2017-07-16:
I wanted to slightly customize how the caps lock led handles things, so I put a file called tada68.c in my keymap folder, tada68/keymaps/maartenwut. Like other files such as Makefile and config.h, I expected it to be included when building the firmware, but it's not.

jackhumbert, 2017-07-16:
Did you add `tada68.c` to the `SRC +=` in your `rules.mk`?

Maartenwut, 2017-07-16:
I did not. I've never seen it before.

fredizzimo, 2017-07-16:
You should not really need to modify things from the keymap files.

Normally keyboards have a `led_set_user`, which you can override in your keymap. The `tada68` doesn't however, I suggest that you look at some examples from other keyboards and fix the real `tada68.c` file to support it and send a pull request.

Maartenwut, 2017-07-16:
This worked. Thanks!

closed 2017-07-16 by Maartenwut