qmk/qmk_firmware #1505: keymap in eeprom support?

YCF, 2017-07-18:
https://github.com/tmk/tmk_keyboard/pull/113

Talljoe, 2017-07-27:
[Wilba's fork](https://github.com/Wilba6582/qmk_firmware) for the Zeal60 has support for this along with a [tool](https://github.com/Wilba6582/zeal60). I would like to see it expanded and backported to QMK.

kalupa, 2017-08-01:
Would be nice to make that tool cross-platform, too, considering QMK itself (mostly) is. I'm going to take a look at the tool itself, since I haven't, and see how easily that could be done. I'm guessing, having not looked at more than the read me, that it's a GUI tool, so that'll make it more difficult.

Talljoe, 2017-08-01:
Right now it's pretty barebones and CLI only. Someone has a fork that builds under Linux and it doesn't look like Mac would be too far away.

The biggest issue is that `QK_` codes change depending on what the #defines are, so those would have to be locked into place for all firmware, or you'd have to build a version of your tool to match the version of your firmware (meaning web configurators wouldn't work).

It's all very exciting, though. Was on a Discord where someone built a Gherkin for someone else, shipped it to them, and got the layout wrong. Now it's being shipped back. This feature would make situations like that much easier to resolve!

drashna, 2019-04-05:
This is being supported by VIA now.

Though they "Dynamic Keymap" feature is a core feature in QMK Firmware

YCF, 2019-04-14:
> This is being supported by VIA now.
> 
> Though they "Dynamic Keymap" feature is a core feature in QMK Firmware

Is VIA Open Source?

drashna, 2019-06-17:
Yes. https://github.com/olivia/via-config

But the secret sauce for VIA is raw HID and Dynamic Keymaps.  With those features, anyone can implement their own method of implementing this.

closed 2022-06-16 by tzarc