qmk/qmk_firmware #1507: Cycling and Directly Choosing Layers with the same key?

galengold, 2017-07-19:
I am creating a macro pad that is 1 row by 6 columns. I intend to have the column 0 key be used to switch layers so I can have different sets of macros. The table below should work so that I can cycle through each layer with a momentary press of the key.

|         | Col 0 | Col 1 | Col 2 | Col 3 | Col 4 | Col 5 |
|---------|-------|-------|-------|-------|-------|-------|
| Layer 4 | TO(0) | M20   | M21   | M22   | M23   | M24   |
| Layer 3 | TO(4) | M15   | M16   | M17   | M18   | M19   |
| Layer 2 | TO(3) | M10   | M11   | M12   | M13   | M14   |
| Layer 1 | TO(2) | M05   | M06   | M07   | M08   | M09   |
| Layer 0 | TO(1) | M00   | M01   | M02   | M03   | M04   |

**Is it possible to have layer cycling *and* use it like a modifier for layer switching when held?** For example: if I press Col0 + Col1 on any layer, it should switch to layer 0 (See the table below).

| Key Combo | Action |
|---------|-------|
| Col0 + Col5 | Go to layer 4 |
| Col0 + Col4 | Go to layer 3 |
| Col0 + Col3 | Go to layer 2 |
| Col0 + Col2 | Go to layer 1 |
| Col0 + Col1 | Go to layer 0 |

galengold, 2017-07-20:
One thought I had was to set the col0 key as the Leader and create leader combinations to go to layers, but that doesn't allow me to cycle through like in the first table.

drashna, 2018-03-25:
Tap dance may actually be perfect for this.  Just use the count to set the layers. 
https://docs.qmk.fm/feature_tap_dance.html

foosinn, 2019-04-09:
did anyone implement this using the tap dance?

foosinn, 2019-04-15:
for anyone interested: i use a currently use a macro to reset the layers to google back to my default:

```c
const macro_t *action_get_macro(keyrecord_t *record, uint8_t id, uint8_t opt) {
  switch(id) {
  case 100:
    layer_clear();
    return MACRO_NONE;
    break;
  default:
    return MACRO_NONE;
    break;
  };
};
```

On the default layer i have `OSL(1)`, on the second `OSL(2)` and on the third `M(100)`.

jackhumbert, 2020-03-13:
This sounds like a good fit - if you have any complications with implementing this, feel free to comment/reopen this.

closed 2020-03-13 by jackhumbert