qmk/qmk_firmware #1509: KanaLock status doesn’t update

Flavien21, 2017-07-20:
Hello,

I use the KanaLock modifier and there is a slight issue with it.

Whenever I press KanaLock, the state of the keyboard does change but the led does not turn on/off.
However, if I press another lock modifier like CapsLock or NumLock, the KanaLock led updates its status.

I first thought of a Windows problem and then downloaded the software TrayStatus to make sure, but in TrayStatus the KanaLock status updates instantaneously.

Is there something wrong with KanaLock in qmk or is it a Windows problem anyways?

Regards,
Flavien

fredizzimo, 2017-07-20:
Which keyboard are you using? It's up to the keyboard to display the correct LED status, and I believe most ignore the KanaLock.

Flavien21, 2017-07-20:
I’m using Ergodox EZ.
I prepared a macro sending KanaLock then twice CapsLock to force update status, haven’t tryed it yet (will do it monday).

Flavien21, 2017-08-11:
Well the workaround macro works.

closed 2017-08-11 by Flavien21