qmk/qmk_firmware #1521: Make LT send the keycode if no other key was tapped

Maartenwut, 2017-07-27:
Right now, the behavior of LT is that it will switch to the layer if it wasn't let go within x milliseconds. It won't send a keycode if the layer changed. Wouldn't it be better if the keycode were to be sent if no other key was pressed?

ylixir, 2017-08-07:
#1359
Documented [here](https://docs.qmk.fm/key_functions.html#permissive-hold).
It's not perfect, i can still get it to trigger on purpose, but it doesn't happen on accident anymore. Might just be a latency issue now, because of my split board.

lambdasakura, 2018-04-25:
I have the same problem.

for macros other than `LT` (eg `ALT_T` and `CTL_T`), keycode could be sent  if no other key was pressed by enabling `RETRO_TAPPING`. However, `RETRO_TAPPING` does not seem to work well with `LT`.