qmk/qmk_firmware #1523: Need clarification on combining multiple toggle and hold layers

rask, 2017-07-28:
I have some layers as such planned:

- `0` base layer with normal alphanums and modifiers
- `1` alternate base layer via `TG` with some keys changed for extended use (e.g. split space setups, some modifiers nulled, etc.)
- `2` via `MO` for number row to F-row and other QOL goodies on 60%
- `3` via `MO` for media controls and macros

Now when I call `MO` on `2` and `3` they should just map `KC_TRANS` and `KC_NO` properly and the changes between `0` and `1` should work normally?

What would happen in the following situation:

1. Initially, I am on `0`
2. Then I toggle on `1`
3. I am on `1`
4. Then I press+hold Fn3 to `3`
4. Now I am on `1` _and_ `3` at the same time
5. While still holding `3` I now toggle off `1`

Am I now on `0` _and_ `3` as expected or will something break?

I know there would be no problems if I release `3` before toggling off `1` but I would rather not have a problem so close to happening.

What would happen if `1` was a `MO` instead of `TG`, any change other than how the layer is activated?

Thanks!

Maartenwut, 2017-07-28:
The best way to find out is to just try.

rask, 2017-07-28:
@Maartenwut otherwise I would, but I'm going to use MX lock switches to toggle `1`, hence my last question about the difference between `MO` and `TG` for this.

I would rather not solder expensive lock switches on just to "try" it out in case I mess up while desoldering or something. ;)

Maybe I could try this out on some other board with a temp firmware build, though.

Maartenwut, 2017-07-28:
A lock switch.. Nice! Use MO or else you'll be pressing 4 times to go back and fourth.

BalzGuenat, 2017-08-12:
@rask I suggest you try with an existing board. You don't actually need a lock switch in your board to test this. Just hold down the switch that would be the lock switch.

rask, 2017-08-28:
@BalzGuenat I'm going to assemble the firmware today, will try with just some pins first and then alter to toggle layers in case the momentary layers end up getting broken in my use. I'll try to remember and post my results here afterward.

rask, 2017-08-29:
It works! The lock switches combined with `MO` layers and such don't break even with random layer jumps. A working example available on my Bananasplit60 keymap: https://github.com/rask/qmk_firmware/tree/rask-keyboards/keyboards/bananasplit/keymaps/rask

I'll close this one.

closed 2017-08-29 by rask