qmk/qmk_firmware #1524: Delay on switching RGB modes inside custom function

Wheels2050, 2017-07-29:
I wrote a custom function for my Fn key that should switch the RGB mode when the key is held down. The code is as follows:

    void action_function(keyrecord_t *record, uint8_t id, uint8_t opt) {
      static uint8_t shift_esc_shift_mask;
      switch (id) {
      /* This is intended to indicate the layer when the appropriate key is pressed.
          There is a delay in the RGB mode switching for some reason, though. */
      case 1:
        {
          static uint8_t currentmode = 1;
          if(record->event.pressed){
          currentmode = rgblight_config.mode;
          rgblight_mode(24);
          layer_on(1);
        } else {
	  rgblight_mode(currentmode);
	  layer_off(1);
         }
       }
     }
    }

However, there seems to be a random delay in the RGB mode switch happening on my KBD75. Sometimes the mode changes instantaneously, while other times it takes a second or so for the mode to switch.

Is this some issue with the timing of the relevant loop in QMK, or should I be doing this some other way?

drashna, 2017-09-29:
IIRC, there is a normal delay for switching due to the "resource intensive nature of the RGB code".

That said,  using the matrix scan may work better, and would detect the layer, regardless of how it got there. 

https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/drashna/keymap.c

drashna, 2017-11-18:
Later on "found" this. 
https://github.com/qmk/qmk_firmware/blob/master/keyboards/ergodox_ez/keymaps/default/keymap.c#L208

The `layer_state_set_user` function lets you call any action on layer change, and works very well for this.  And this may help with the speed/performance issue.

closed 2018-10-21 by drashna