qmk/qmk_firmware #1535: Operation not permitted - /usr/local/opt/avr-gcc/avr

Miodec, 2017-08-02:
Hi, i cant brew install avr-libc

`~> brew install avr-libc
==> Installing avr-libc from osx-cross/avr
==> Using the sandbox
==> Downloading https://download.savannah.gnu.org/releases/avr-libc/avr-libc-2.0
Already downloaded: /Users/jacekbartnik/Library/Caches/Homebrew/avr-libc-2.0.0.tar.bz2
==> ./configure --build=i686-apple-darwin16.6.0 --prefix=/usr/local/Cellar/avr-l
==> make install
==> copying /usr/local/Cellar/avr-libc/2.0.0/avr -> /usr/local/opt/avr-gcc
Error: Operation not permitted - /usr/local/opt/avr-gcc/avr`

Maartenwut, 2017-08-02:
Try running it as root?

Miodec, 2017-08-02:
Brew cant be ran as root

[image]

Maartenwut, 2017-08-02:
What about `brew doctor`?

Miodec, 2017-08-02:
Tried it already
[image]

skullydazed, 2017-08-02:
Does anyone besides you use that machine? It's possible you have conflicting permissions. You can use this to force everything over to your user:

`sudo chown -R $USER /usr/local`

Be warned that if you have mysql (or certain other server packages installed) that may break them.

Miodec, 2017-08-02:
Didnt work 
[image]

nooges, 2017-08-03:
After talking to @Miodec, turns out the issue is due to Homebrew now turning on sandboxing for all non-homebrew taps, which prevents the final copy operation from succeeding.

I've added a comment to this issue with the homebrew-avr tap: https://github.com/osx-cross/homebrew-avr/issues/19

nooges, 2017-08-04:
This should be resolved now that https://github.com/osx-cross/homebrew-avr/pull/54 has been merged, which closes https://github.com/osx-cross/homebrew-avr/issues/19.

closed 2017-10-22 by skullydazed