qmk/qmk_firmware #1544: More than 2 dynamic macros

zfjagann, 2017-08-04:
### TLDR

How hard would it be to program more than 2 dynamic macros?

### Background

One of the things I miss most from my pok3r was the ability to rebind keys on the fly. Usually I would rebind short macros of 2 or 3 characters that I had to type repeatedly for specific cases during whatever game I was playing. It was easy to pause the game, reprogram a macro, then resume and use the bound macro a few times, and then I could clear it again. It made repetitive tasks very easy.

### More than 2 dynamic macros with QMK?

I'd like to do something similar with QMK, where I can bind several macros on the fly.

I know that's not possible out-of-the-box, but I'm wondering how difficult this would be to add, and how I could go about doing it.

I don't really need that many, but 2 is definitely not enough. I figure if it's limited to 128 keypresses, then splitting that into 8 macros would be 16 keypresses, which is a ton for what I want to do. Or 16 macros at 8 each would be suitable too.

I'm wondering how I might go about allowing this using QMK. Is this easy or hard? Where should I get started? Does something like this exist already?

jayliu50, 2017-09-07:
Would be great to have, for sure. From what I understood about the code, it's implemented by cutting a buffer in half and filling it up from the middle of the buffer; that is why there can only be two. Adding more is theoretically possible, but I think would be a more disruptive rewrite. Perhaps someone who knows better could clarify what I have said.

drashna, 2018-03-25:
I believe that the issue here is that there is a limit on the RAM that many of the controllers have. As well as eeprom size. 

So adding more may not be feasible for all boards.

@jackhumbert, @fredizzimo, or the like would have to confirm this, though.

fredizzimo, 2018-04-07:
This would need a completely new implementation. It wouldn't be that hard to make, but still someone has to do it. 

As @drashna mentioned, the EEPROM size is an issue, but still we could fit quite many small ones there.  Currently a single configurble buffer, which defaults to 128 is used. But that could also be for example eight 16 byte buffers, or 16 eight byte buffers. At the moment eveything is stored in RAM, which is very limited, but if the storage is changed to EEPROM, then almost 1024 bytes should be available.

stash, 2020-09-08:
I looked at the code recently and I can say it's definitely possible. The main constraint is that memory is allocated statically. There's also a bunch of code to change, but it's far from impossible, just a fair amount of work, IMO:
* refactor the current `static` variables in `process_dynamic_macro` to a single struct that defines a pair of buffers
* Add a `#define DYNAMIC_MACRO_SLOT_PAIRS N` so that you end up with 2*N macro slots
* Do some CPP magic to define new `DYN_REC_START` and `DYN_MACRO_PLAY` keycodes for each pair. Unfortunately, right now, there aren't ranges reserved for this in `quantum_keycodes.h`.
* Change the main logic to use range checks instead of switch blocks. Probably easiest if all the `DYN_REC_START*` codes are all contiguous, same for `DYN_MACRO_PLAY`

The alternative would be to dynamically allocate memory somehow for each slot, but I am not aware how viable that is in QMK. Maybe a linked list of recorded key events. What happens when memory can't be allocated during a recording?

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc