qmk/qmk_firmware #1563: Trouble with send_string

rogerhi, 2017-08-09:
I have a keymap on a chinese gh60 using qmk satan that I am trying to compile and I cant get it to work:

Error:
```
/mnt/c/unixdump/qmk-satan-rgb-master (1)/qmk-satan-rgb-master/keyboard/satan/keymaps/keymap_yang.c:124: undefined reference to `SEND_STRING'
collect2: error: ld returned 1 exit status
```

Code looks like:
```
 case 1: // this would trigger when you hit a key mapped as M(1)
      if (record->event.pressed) {
        SEND_STRING ("QMK is the best thing ever!");
      }
      return false;
      break;
```

I can only assume it needs something included or something but I can't work out from the doco what it would be. 
Any help most appreciated.

drashna, 2018-03-25:
Is this still happening?

drashna, 2018-10-21:
Sounds like probably not.

If so, rebase your repo, and see if that fixes it. 

And if that doesn't help, feel free to reopen.

closed 2018-10-21 by drashna