qmk/qmk_firmware #1568: SHIFT/CTRL + GRAVE_ESC not released correctly if SHIFT/CTRL is released first

BalzGuenat, 2017-08-11:
To reproduce:
1. Press and hold GUI or SHIFT.
2. Press and hold GRAVE_ESC.
3. Release GUI or SHIFT.
4. Release GRAVE_ESC.

This will spam `\`` until the next keypress.

This is because when the ESC_GRAVE key is released, only the currently pressed mods are used to determine which keycode (GRAVE or ESC) gets released.

This looks like an issue that would arise with layers as well. We should look at how it's solved there.

BalzGuenat, 2017-08-12:
I have looked into how this issue is handled for layers.

The function that is responsible for finding the right keycode to release is [`store_or_get_action`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action_layer.c#L166) in [`action_layer.c`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action_layer.c). Looks like we store in [`source_layers_cache`](https://github.com/qmk/qmk_firmware/blob/master/tmk_core/common/action_layer.c#L126) for every physical key the layer on which it was pressed (albeit in a complicated encoding). When a key is released, we look up the layer on which it was originally pressed, get the action of that key on that layer and run `process_action` with that action.

Since `GRAVE_ESC` does not work with normal layers, it's not possible to integrate the handling of this issue with the above mentioned `source_layers_cache`. A close analogue is also impossible, since `GRAVE_KEY` is not a physical key but a keycode.

I suggest we store the generated keycode when `GRAVE_ESC` is pressed so we can release the correct one once it's released. I will create a pull request for this shortly. If you disagree with the basic idea to solve this, we should discuss that here. If you just have comments concerning the particular implementation, put those on the pull request.

drashna, 2018-03-25:
This should be fixed. 

Closing.  If it's not, reopen or open a new issue, please!

closed 2018-03-25 by drashna