qmk/qmk_firmware #1578: S60-X firmware doesn't work on original S60-X

Daniel-Khodabakhsh, 2017-08-13:
The S60-X firmwares (s60_x-default-*) do not work on the original S60-X.
Uploading the hex leads to the topmost row of keys being unresponsive (ESC to Backspace).

This problem doesn't occur with the [VinnyCordeiro's tmk_keyboard/s60-x](https://github.com/VinnyCordeiro/tmk_keyboard/tree/master/keyboard/s60-x) version of the firmware.

Dbroqua, 2017-08-13:
Hello,

Do you have RGB version of the S60-X or other version?

Daniel-Khodabakhsh, 2017-08-13:
Hey! I have the non-RGB version, the first one on Massdrop I believe. (Before the S60-X-RGB was available).

Dbroqua, 2017-08-13:
It's strange,

pins for cols and rows are the same between your link and the default case in QMK.

Daniel-Khodabakhsh, 2017-08-13:
Humm, do you think it's due to the warning I'm getting when running `make s60_x` ?

I'm also using fedora 25's repo version of `dfu-programmer` to flash, which is at version `0.6.2-5.fc24`.

`[dan@localhost qmk_firmware]$ make s60_x
QMK Firmware v0.5.99
WARNING:
 Some git sub-modules are out of date or modified, please consider runnning:
 git submodule sync --recursive
 git submodule update --init --recursive

 You can ignore this warning if you are not compiling any ChibiOS keyboards,
 or if you have modified the ChibiOS libraries yourself. 

Making s60_x/default with keymap ansi_qwertz                                                           [OK]
Making s60_x/default with keymap bluebear                                                              [OK]
Making s60_x/default with keymap custom                                                                [OK]
Making s60_x/default with keymap dbroqua                                                               [OK]
Making s60_x/default with keymap default                                                               [OK]
Making s60_x/default with keymap hasu                                                                  [OK]
Making s60_x/default with keymap hhkb                                                                  [OK]
Making s60_x/default with keymap iso                                                                   [OK]
Making s60_x/default with keymap jpec                                                                  [OK]
Making s60_x/default with keymap plain                                                                 [OK]
Making s60_x/default with keymap poker                                                                 [OK]
Making s60_x/default with keymap poker_bit                                                             [OK]
Making s60_x/default with keymap poker_set                                                             [OK]
Making s60_x/default with keymap spacefn                                                               [OK]`

Daniel-Khodabakhsh, 2017-08-13:
The hex file I'm using is `s60_x_default_default.hex`.

Dbroqua, 2017-08-14:
all seems ok.

jackhumbert, 2017-08-15:
This may be due to the [backlight being set as B7](https://github.com/qmk/qmk_firmware/blob/master/keyboards/s60_x/config.h#L36) - it's not enable for the S60-X project, but it is for the `default` subproject. Do you happen to know what the actual backlight pin is? If there isn't one, it should be [disabled in the subproject](https://github.com/qmk/qmk_firmware/blob/master/keyboards/s60_x/default/rules.mk#L4).

Daniel-Khodabakhsh, 2017-08-16:
Thanks Jack, There's [no backlight or any LED](https://lh3.googleusercontent.com/-WHbzvUpEOy8/VpD83ASCoHI/AAAAAAAAA48/tDH4ELYQIXo/w1190-h797/2016-01-09_13-25-04.jpg) so I'll try that out and let you know if it works.

Daniel-Khodabakhsh, 2017-08-19:
[Changing BACKLIGHT_ENABLE to 'no'](https://github.com/qmk/qmk_firmware/blob/master/keyboards/s60_x/default/rules.mk#L4) breaks the `make s60_x` command. Upon executing this command, only the `s60_x_default_ansi_qwertz.hex` file is generated (even though the output is the same as I mentioned above, with [OK] for each keymap).
No error messages are outputted either.

Daniel-Khodabakhsh, 2017-08-20:
Tried to clean my make environment, created a new layout triggered a more thorough `make s60_x` execution and I was able to generate a hex file.

Setting BACKLIGHT_ENABLE to no fixed it! The first row is now working!

closed 2017-08-20 by Daniel-Khodabakhsh