qmk/qmk_firmware #1588: Keyboard Readme Refresh

skullydazed, 2017-08-17:
This is a master issue to track the ongoing work to update keyboard `readme.md` files. These keyboards need to be updated still:

- [ ] atomic
- [ ] chibios_test
- [ ] converter
- [ ] ergodox
- [ ] kc60
- [ ] kinesis
- [ ] kmac
- [ ] lets_split
- [ ] m10a
- [ ] maxipad
- [ ] nyquist
- [ ] orthodox
- [ ] ps2avrGB
- [ ] subatomic

This is a sample of the readme format:

---

```md
Planck
===

![Planck](http://i.imgur.com/q2M3uEU.jpg)

A compact 40% (12x4) ortholinear keyboard kit made and sold by OLKB and Massdrop. [More info on qmk.fm](http://qmk.fm/planck/)

Keyboard Maintainer: [Jack Humbert](https://github.com/jackhumbert)
Hardware Supported: Planck PCB rev1, rev2, rev3, rev4, Teensy 2.0
Hardware Availability: [OLKB.com](https://olkb.com), [Massdrop](https://www.massdrop.com/buy/planck-mechanical-keyboard?mode=guest_open)

Make example for this keyboard (after setting up your build environment):

    make planck-rev4-default

See [build environment setup](https://docs.qmk.fm/build_environment_setup.html) then the [make instructions](https://docs.qmk.fm/make_instructions.html) for more information.
```

nooges, 2017-08-18:
There needs to be blank lines in between "Keyboard Maintainer", "HW Supposed", etc., otherwise, the lines get mushed together when the Markdown parser is run on it, like this:

[image]

Or add bullet points to the beginning of each of those lines.

jackhumbert, 2017-08-18:
Ah, thanks! I think I was planning on noting that there should be two spaces at the end of each line, but it slipped my mind :)

ryanmaclean, 2017-08-19:
Can you look at your line endings for that default readme? Currently there aren't any, and they display as one line on github.
Perhaps a table would be best?

jackhumbert, 2017-08-19:
Yeah, the ones that were changed previously need those spaces added in - I've only just added them to the example here, not the actual files!

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc