qmk/qmk_firmware #1600: Moving split keyboard files to common location

nooges, 2017-08-18:
With the addition of various split keyboards lately based off of the Let's Split keyboard, there's been a lot of duplication of the files that handle keymap matrix and communication between halves. However, if there's a change to the files for one split keyboard that could be useful to others, the change doesn't get propagated unless done so by the individual maintainers.

Example issue and PR that's only been applied to the Let's Split and Nyquist so far: 
https://github.com/qmk/qmk_firmware/issues/1511
https://github.com/qmk/qmk_firmware/pull/1519

Also this one regarding EEPROM handedness: https://github.com/qmk/qmk_firmware/pull/1576

I'm opening up this issue to gather some feedback on refactoring these common split keyboard files and where they should go to make it more maintainable and easier for someone to create a new board.

Some of the files that are common:
- [ ] eeprom-lefthand.eep
- [ ] eeprom-righthand.eep
- [ ] i2c.c
- [ ] i2c.h
- [ ] matrix.c
- [ ] serial.c
- [ ] serial.h
- [ ] split_util.c
- [ ] split_util.h

Tagging: @That-Canadian

jackhumbert, 2017-08-18:
Thanks for bringing this up! This has been on my list of things to do since creating `drivers/` (where these files should live), but there were enough differences between the files listed that I wasn't 100% comfortable merging all of them, especially since these are now production boards (whereas it was pretty much just the DIY lets_split before).

I think the biggest challenge, and something I'd definitely like to see is a smarter matrix file that can be used by split keyboards in the same way that regular boards use the `config.h` file to declare their matrices. Supporting both the M/S atmega32u4 chips, but also IO expanders would be important here. I would hope that boards like the Ergodox would be able to use this as well.

That-Canadian, 2017-08-19:
So I'm down to help with this as much as time and skill allows. The I2C code definitely needs some cleanup so it's not so "hacky", it needs a proper API / interface written so we can simply send info master -> salve and vice versa. Same goes for the serial implementation. It's been a while since I've officially dabbled in I2C but I'm willing to lend my help, I know Danny has been messing with the I2C code in his spare time.

The EEPROM changes are easy to move over, it doesn't affect any of the core I2C or serial code.

I'll also poke around the existing matrix code and see if I can help there. Unfortunately I do not have a board with an IO expander to test with, only a MiniDox, Let's Split and Nyquist.

ahtn, 2017-08-19:
> .The I2C code definitely needs some cleanup so it's not so "hacky", it needs a proper API / interface written so we can simply send info master -> salve 

I had a simpler interface [here](https://github.com/ahtn/tmk_keyboard/blob/1890990eead0dda8b461f6cc702b4610ea9e45a2/keyboard/split_keyboard/matrix.c#L124) that never ended up in QMK.

>  Same goes for the serial implementation

The `serial_slave_buffer` and `serial_master_buffer` are shared between both devices when `serial_update_buffers()` is called. The `serial_slave_buffer` is written to by the slave and is readable by the master. The `serial_master_buffer` is written by the master and is read by the slave. So if you want to control the slave you can just write to `serial_master_buffer`, and the slave can just use the data by reading `serial_master_buffer`. It's not great for sending large amounts of data, but it's good enough for controlling indicator LED's which is all I ever intended to do with it.

drashna, 2018-10-21:
Closing, since this was implemented.

closed 2018-10-21 by drashna