qmk/qmk_firmware #1602: When using a USB-USB with a CODE keyboard, the backlights flash whenever a layer key is pressed.

BalzGuenat, 2017-08-18:
This is with the Pro Micro variant of the converter. Not sure if it exists with other variants.

By "flash", I mean they turn off for a split second before turning on again. Happens on both key press and release.

Unsure what would cause this behavior of the CODE. It doesn't use a driver to manage the backlight, only on-keyboard controls.

If it's not something in the code, I suspect a sudden voltage drop because the chip has to do more work for layer switching. Would need an oscilloscope to measure. Alternatively, hook up power supply for keyboard separately from converter.

BalzGuenat, 2017-08-19:
Could not get separate power supply to work at all. Keystrokes do not reach PC. Unsure why. Will look at code next.

drashna, 2018-03-25:
Were you able to get this resolved?

BalzGuenat, 2018-05-24:
@drashna Sadly no. I also sold the CODE so I'm unable to explore this further.

closed 2018-10-21 by drashna