qmk/qmk_firmware #1620: [Ask] Backlighting on XD75RE

kimkimkere, 2017-08-21:
Hello,

I'm using xd75re with qmk. On every keymap, I see the backlight is turned off, and when I activate the option, I got this error : 

`Compiling: keyboards/xd75/xd75.c [OK]
Compiling: keyboards/xd75/keymaps/kimkim/keymap.c [OK]
Compiling: quantum/quantum.cquantum/quantum.c: In function ‘__vector_17’:
quantum/quantum.c:992:5: error: ‘OCR1x’ undeclared (first use in this function)
     OCR1x = (uint16_t)(((uint16_t)pgm_read_byte(&breathing_table[local_index]) * 257)) >> breath_intensity;
     ^
quantum/quantum.c:992:5: note: each undeclared identifier is reported only once for each function it appears in
 [ERRORS]

tmk_core/rules.mk:358: recipe for target '.build/obj_xd75_kimkim/quantum/quantum.o' failed
make[1]: *** [.build/obj_xd75_kimkim/quantum/quantum.o] Error 1
Make finished with errors
../../../../Makefile:503: recipe for target 'xd75-kimkim' failed
make: *** [xd75-kimkim] Error 1`

So, is anything wrong with that?

Thank you

Ptomerty, 2017-08-21:
Did you define `BACKLIGHT_PIN`? It should be set here: https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L667

kimkimkere, 2017-08-22:
@Ptomerty xd75re's `BACKLIGHT_PIN` is in F5, so it's `NO_BACKLIGHT_CLOCK`. But when I disable `BACKLIGHT_BREATHING`, the backlight is still turned off.

Any suggestion?

Ptomerty, 2017-08-22:
I'm not sure, sorry.

kimkimkere, 2017-08-29:
I think the problem was reversed LED pin. Now the backlight is on.
And maybe I'll make a pull request for proper backlight level.

Thanks

closed 2017-08-29 by kimkimkere

Torreip, 2017-09-23:
@kimkimkere, I have the same issue. What do you mean by "reversed LED pin"? `BACKLIGHT_PIN` should not be F5?