qmk/qmk_firmware #1623: Defining multiple Tap Dance actions results in only the final action functioning.

Ptomerty, 2017-08-21:
Extension of https://github.com/qmk/qmk_firmware/issues/1552:

Reproducible with example code below.

When defining multiple tap dance actions, only the *final* action called will register. In the example code, both tap dance functions end up calling `TD(TD_LAYER_1)`, despite no keys being assigned to that action.

Sample code:
```C
enum {
  TD_LAYER_3 = 0,
  TD_DOUBLE = 0,
  TD_LAYER_1 = 0
};

qk_tap_dance_action_t tap_dance_actions[] = {
  //Tap once for Esc, twice for Caps Lock
  [TD_LAYER_3]  = ACTION_TAP_DANCE_DUAL_ROLE(KC_ESC, 3),
  [TD_DOUBLE] = ACTION_TAP_DANCE_DOUBLE(KC_1, KC_2),
  [TD_LAYER_1]  = ACTION_TAP_DANCE_DUAL_ROLE(KC_ESC, 1)
// Other declarations would go here, separated by commas, if you have them
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

    KEYMAP(
        TD(TD_LAYER_3), TO(1), TD(TD_DOUBLE), KC_MPRV, KC_MPLY, KC_MNXT,
        KC_P7, KC_P8, KC_P9, KC_MUTE, KC_VOLD, KC_VOLU,
        KC_P4, KC_P5, KC_P6, XXXXXXX, KC_UP, XXXXXXX,
        KC_P1, KC_P2, KC_P3, KC_LEFT, KC_DOWN, KC_RGHT,
        KC_P0, KC_PDOT, KC_ENT, KC_SPC),

    KEYMAP(
        KC_ESC, KC_1, KC_2, KC_3, KC_Z, KC_X,
        KC_INS, KC_HOME, KC_PGUP, XXXXXXX, KC_Z, KC_X,
        KC_DEL, KC_END, KC_PGDN, XXXXXXX, KC_Z, KC_X,
        XXXXXXX, XXXXXXX, KC_Z, KC_X, KC_Z, KC_X,
        TO(3), MO(2), XXXXXXX, KC_SPC),
<snip>
```

algernon, 2017-08-21:
Well, this is expected behaviour, because of this part of your code:

```c
enum {
  TD_LAYER_3 = 0,
  TD_DOUBLE = 0,
  TD_LAYER_1 = 0
};
```

All of these symbols will have the value of `0`, so when tap-dance sees any of these keys, it sees `TD(0)` for all three. It has no way of knowing you gave them different names. If you want different functionality, different actions for different keys, you'll have to use different numbers.

If you drop the `= 0` stuff from the enum, you should be fine, I think.

Ptomerty, 2017-08-21:
Ahaha I didn't understand what the enum was for, I feel stupid now. Thanks!

closed 2017-08-21 by Ptomerty