qmk/qmk_firmware #1632: Bluetooth Planck Does Not Transmit Keypresses

dwrz, 2017-08-23:
I have a bluetooth Planck that uses the Adafruit Bluefruit.

With the latest firmware, bluetooth does not work.

I am able to pair and connect to the Bluefruit (checked on three devices). But keypresses are not transmitted.

Keymap is [piemod](https://github.com/qmk/qmk_firmware/tree/master/keyboards/planck/keymaps/piemod). I have ```BLUETOOTH_ENABLE = yes``` in the [Makefile](https://github.com/qmk/qmk_firmware/blob/master/keyboards/planck/keymaps/piemod/Makefile).

The keyboard works fine over cable / USB.

**Expected Behavior**
After pairing and connection, a bluetooth Planck transmits keypresses to the connected device.

**Actual Behavior**
With the latest firmware, bluetooth pairs and connects, but keypresses are not transmitted.

depadiernos, 2017-09-29:
Are you using the 32u4 Feather BLE or the Bluefruit EZ-key? If it's the feather, you could try the following:

```BLUETOOTH_ENABLE = no
BLUETOOTH = AdafruitBLE```

dwrz, 2017-09-29:
Alas, I'm using the Bluefruit EZ-key. I'm using forked code at the moment, but would love to get the BT_ON / BT_OFF features in the newer firmware.

drashna, 2018-10-21:
Was this still an issue?

throwyourboatsgently, 2020-01-09:
I just finished building a Planck Rev 4 that I've had for a few years collecting dust. I'm haaving the same issue with bluetooth not transmitting any keypresses. Has anyone found a solution to this problem?

jceb, 2020-12-21:
I also ran into the same issue with the Adafruit Feather 32u4 BLE. I used the latest development branch, had NKRO disabled and couldn't get any key presses out of the controller. With the regular F_CPU value nothing happened at all, when I reduced it to `8000000` the keyboard started emmitting weird commands upon connecting it via USB but then did't react to key presses.
The keyboard is fine as I tested with a regular pro micro.

tzarc, 2022-06-18:
Closing due to inactivity.

closed 2022-06-18 by tzarc