qmk/qmk_firmware #1633: Incremental encoder support in QMK

EquilibriumDev, 2017-08-24:
Not really an issue, but I raised the subject of encoder support in QMK on Reddit recently, and would like to hear about your philosophy/opinions on some aspects of it so I can try and make a worthwhile contribution.

I am attaching two code files, both for Teensy++ 2.0, proofs of concept for incremental encoder use. One uses interrupts, the other doesn't.

My plan is to use macros and global variables similar to BACKLIGHT_PIN, and configure the pins and interrupts based on this. It could become slightly complex to accommodate interrupt/half-interrupt/non-interrupt modes on all MCs and translate pin selection to interrupt register values, but it's possible and leads me to the questions:

 - I can't find any usages of external interrupts in QMK, only timers and USB interrupts. Do you have any thoughts about using external interrupts? Or are you open to experimentation?
 - Would it be okay to restrict users to full-interrupt mode and limit the pins to interrupt-only for example?
 - What is the smallest set of MCs for which support is required to merge? Not sure how much I can test on.

<br>

Non-interrupt encoder support:
```c
/*
* Proof of concept for incremental encoder library without using interrupts.
* Breadboarding:
*    - a Teensy++ 2.0
*    - one LED-resistor pair on C7, and one on B7.
*    - one incremental rotary encoder, wired this way:
*
*                   E7
*   ___             |     _______
*  |   |------------*----|__10k__|----- +5V
*  | O |-- GND            _______
*  |___|------------*----|__10k__|----- +5V
*                   |
*                   E6
*
* The goal is to illuminate the LED on B7 once per clockwise detent-to-detent movement,
* and the LED on C7 once per counter-clockwise detent-to-detent movement.
*/

#include <avr/io.h>
#include <avr/pgmspace.h>
#include <util/delay.h>

//#if defined(__AVR_ATmega32U4__) || defined(__AVR_AT90USB1286__)
#define LED_ON		(PORTC |= (1<<7))
#define LED_OFF		(PORTC &= ~(1<<7))
#define LED_2_ON		(PORTB |= (1<<7))
#define LED_2_OFF		(PORTB &= ~(1<<7))
#define LED_CONFIG	(DDRC |= (1<<7))
#define LED_2_CONFIG	(DDRB |= (1<<7))
#define CPU_PRESCALE(n)	(CLKPR = 0x80, CLKPR = (n))

int main(void)
{
	// set for 16 MHz clock, and make sure the LED is off
  CPU_PRESCALE(0);
  LED_CONFIG;
  LED_2_CONFIG;

  LED_OFF;
  LED_2_OFF;

  // Both pins used for the encoder are inputs.
  DDRE &= ~(3 << 6); // 11000000
  PORTE &= ~(3 << 6); // 11000000

  int is_moving_ccw = 0;
  int is_moving_cw = 0;
  int previous_status = 0;
	while (1) {
    int pin_status = (PINE & (3 << 6)) >> 6;

    // Possible changes:
    // 3 <-> 1 <-> 0 <-> 2 <-> 3 (cw)
    // 3 <-> 2 <-> 0 <-> 1 <-> 3 (ccw)
    switch (pin_status) {
    case 0:
      // Both pins are 0
      if (!is_moving_ccw && !is_moving_cw) {
        // Should not happen.
      }
      break;
    case 1:
      // Pin 1 only
      if (!is_moving_ccw && !is_moving_cw) {
        is_moving_ccw = 1;
      }
      break;
    case 2:
      // Pin 2 only
      if (!is_moving_ccw && !is_moving_cw) {
        is_moving_cw = 1;
      }
      break;
    case 3:
      // Both pins up, default state
      if (is_moving_ccw && previous_status == 2) {
        LED_ON;
        _delay_ms(10);
        LED_OFF;
      }
      else if (is_moving_cw && previous_status == 1) {
        LED_2_ON;
        _delay_ms(10);
        LED_2_OFF;
      }
      is_moving_ccw = 0;
      is_moving_cw = 0;
      break;
    }

    previous_status = pin_status;
  }
}
```

<br> 

Interrupt encoder support:
```c
/* 
 * Proof of concept for incremental encoder library using interrupts.
 * Breadboarding:
 *    - a Teensy++ 2.0
 *    - one LED-resistor pair on C7, and one on B7.
 *    - one incremental rotary encoder, wired this way:
 *         
 *                   E7 (INT7)
 *   ___             |     _______
 *  |   |------------*----|__10k__|----- +5V
 *  | O |-- GND            _______
 *  |___|------------*----|__10k__|----- +5V
 *                   |
 *                   E6 (INT6)
 *
 * The goal is to illuminate the LED on B7 once per clockwise detent-to-detent movement,
 * and the LED on C7 once per counter-clockwise detent-to-detent movement.
 */

#include <avr/interrupt.h>
#include <avr/io.h>
#include <avr/pgmspace.h>
#include <util/delay.h>

//#if defined(__AVR_ATmega32U4__) || defined(__AVR_AT90USB1286__)
#define LED_ON		(PORTC |= (1<<7))
#define LED_OFF		(PORTC &= ~(1<<7))
#define LED_2_ON		(PORTB |= (1<<7))
#define LED_2_OFF		(PORTB &= ~(1<<7))
#define LED_CONFIG	(DDRC |= (1<<7))
#define LED_2_CONFIG	(DDRB |= (1<<7))
#define CPU_PRESCALE(n)	(CLKPR = 0x80, CLKPR = (n))

volatile int pin_a_event = 0;
volatile int pin_b_event = 0;

ISR(INT6_vect)
{
  pin_a_event = 1;
}

ISR(INT7_vect)
{
  pin_b_event = 1;
}

int is_moving_ccw = 0;
int is_moving_cw = 0;
int previous_status = 0;
int pin_status = 3;

static inline void update_leds(void) {
  // Possible changes:
  // 3 <-> 1 <-> 0 <-> 2 <-> 3 (cw)
  // 3 <-> 2 <-> 0 <-> 1 <-> 3 (ccw)
  switch (pin_status) {
  case 0:
    // Both pins are 0
    if (!is_moving_ccw && !is_moving_cw) {
      // Should not happen.
    }
    break;
  case 1:
    // Pin 1 only
    if (!is_moving_ccw && !is_moving_cw) {
      is_moving_ccw = 1;
    }
    break;
  case 2:
    // Pin 2 only
    if (!is_moving_ccw && !is_moving_cw) {
      is_moving_cw = 1;
    }
    break;
  case 3:
    // Both pins up, default state
    if (is_moving_ccw && previous_status == 2) {
      LED_ON;
      _delay_ms(10);
      LED_OFF;
    }
    else if (is_moving_cw && previous_status == 1) {
      LED_2_ON;
      _delay_ms(10);
      LED_2_OFF;
    }
    is_moving_ccw = 0;
    is_moving_cw = 0;
    break;
  }

  previous_status = pin_status;
}

int main(void)
{
	// set for 16 MHz clock, and make sure the LED is off
  CPU_PRESCALE(0);
  LED_CONFIG;
  LED_2_CONFIG;

  LED_OFF;
  LED_2_OFF;

  DDRE &= ~(3 << 6); // Initialize the E6 and E7 pins as input.
  PORTE &= ~(3 << 6); // 11000000

  cli();

  SREG = 0b00000000; // Not sure this is useful
  EIMSK = 0xC0; // Enable External Interrupts INT6 and INT7
  EIFR = 0xC0;
  EICRB = 0x50;  // Interrupt condition for INT7:4 (INT3:0 is setup with EICRA)
                 // 01 for one pair of bits means any change triggers interrupt.
  sei();

  while (1) {
    if (pin_a_event) {
      cli();
      pin_status = (PINE & (3 << 6)) >> 6;
      pin_a_event = 0;
      update_leds();
      sei();
    }
    if (pin_b_event) {
      cli();
      pin_status = (PINE & (3 << 6)) >> 6;
      pin_b_event = 0;
      update_leds();
      sei();
    }
  }
}
```

Wilba6582, 2017-08-24:
I plan to use rotary encoders in future, and connect them to the switch matrix. This rules out using interrupts. There's probably no advantage to using interrupts for this, as switch matrix scanning happens fast enough for rotary encoder pin state detection. Ideally, encoder handling in QMK core code would allow the encoder's pin state to come from unused row/col locations in the switch matrix... i.e. after scanning the matrix, route the encoder pin states to a generalized encoder handling routine. Then perhaps QMK has callbacks into the keyboard/user level with events, i.e. which encoder changed and the direction, so the user can define what should happen (i.e. modify an internal variable, send out keycodes, etc.)

EquilibriumDev, 2017-08-24:
I had ruled out pin state detection because I didn't think the matrix
scanning loop would be fast enough, how fast is it?
If we can safely rely on pin state, then I agree that make it part of the
matrix would be an elegant solution, although I don't think we can simply
treat it as a switch unfortunately (not sure we can add the encoder
handling code transparently). I don't know whether encoders work with only
one pin up, do you know?

The last part of your comment follows what I was envisioning, with or
without interrupts.

EquilibriumDev, 2017-08-24:
Actually never mind, I forgot that one could define custom keycodes, I suppose the encoder code can be done there...
Still wondering about the loop period...

When were you planning to start using encoders for your code?

jackhumbert, 2017-08-24:
The ones I implemented previously were non-interrupt-based, and I think that's probably the way to go with these, but I'm open to interrupt-based ones as well.

We're looking at using encoders that also press down - that part would be implemented into the normal matrix, but we're going to have a separate system for the encoders. In the software, it'd still run with the matrix_scan. I can see the point in trying to use them all in one physical matrix, but I'm not sure it's worth it - we're trying to allow for a couple different positions with ours.

In the non-interrupt version, I think it'd be pretty easy to support all MCUs/pins we support currently - the quantum matrix file does something similar for AVR (I don't think this has been done for ARM yet).

Wilba6582, 2017-08-24:
@EquilibriumDev I wasn't suggesting using the keymap to define where the encoders were connected... those locations in the matrix would just not be used... you would have to mask them out somehow, so that the core matrix handling doesn't see encoder state changes as keypresses.

TBH I was just adding my two cents to the discussion... that it would be a "nice to have" if encoder handing in QMK was generic, so you could route changes in the switch matrix to it, as a custom matrix handler at the keyboard level. This means people can add encoders to existing keyboard PCBs (or hand-wired builds) by using the I/O pins already allocated to the switch matrix, or spare I/O pins. For example, many 60% PCBs have empty locations in the switch matrix between mods and spacebar, in theory, you could manually wire in a rotary encoder or two ;-)

EquilibriumDev, 2017-08-24:
Yes, my apologies @Wilba6582 I only fully understood what you read after, and I kinda thought out loud (in text...) :)
I think what you and Jack are looking at is actually the easiest way to handle encoders (I like it!). Basically define the pins, and run the handling logic in matrix_scan. And we can use a spare or matrix-assigned I/O pin, I hadn't considered that.
I wonder how one would handle the +5V needed for the encoder, we don't want it on during the matrix_scan, I guess we should use an I/O pin to power it on when needed, so we would need 2N+1 pins for N encoders.

I can give it a try, put the logic in drivers/. Would you be okay with this?

I agree that the push-button on some encoders can be thought of as another switch, that's actually pretty cool, for a "mute" command for example, or "backlight off" on another layer.

jackhumbert, 2017-08-24:
Sure! If you stick the implementation in `drivers/`, then call the scan from [here](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L643), that should be good. 

Check out [lines like these](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum.c#L690) for how to use arbitrary pins (a la `#define ENCODER_PIN_1 B5`) for AVR - I don't think we have any examples for ARM yet.

Wilba6582, 2017-08-24:
@EquilibriumDev you don't need to add the +5V and resistors... the I/O pins have internal pull-up resistors, so you should learn how to enable these in code. Your demo schematic would work fine, connecting the encoder's common pin to ground. If you want to wire this into a matrix, then you would connect the encoder's common pin to the matrix row pin and the encoder's A and B pins to matrix column pins (assuming the matrix is wired as "select row, read columns"). Also for anti-ghosting, you can add diodes in the same direction as already present in the switch matrix, e.g. between I/O pin and encoder A and B pin, cathode pointing to encoder, thus the encoder "sinks" the current coming from the I/O pin, pulling the I/O pin's voltage to 0V when the encoder's common pin is 0V because that row in the matrix was selected (the row's I/O pin was 0V, the others high-impedance i.e. not sinking any current).

EquilibriumDev, 2017-08-24:
I had forgotten about pull-up resistors on pins, it definitely works! I will update my example code to use them.
I will need some time to think about your proposed wiring and the anti-ghosting solution. Not that I doubt, but while I code a lot, this is the first time I'm trying something like this, especially on a Teensy. Wiring the common pin to the ground would have been the easy solution, but I like a challenge.

I realized with your comment that, if one selects pins used in the matrix, they have to take ones that are read, not written, unless it's fine to switch a pin's role repeatedly...

Thanks for your patience, both of you!

Wilba6582, 2017-08-24:
If you don't want to wire this into a matrix (and write the code), you don't have to... if you have two spare I/O pins, just do it as you showed in your schematic. I only mention how it could be done in a matrix in case you were interested. AFAIK the "ideal" way to connect an encoder into a switch matrix is to put the common pin on what will be "selected" by the matrix, then "read off" the two states. An encoder is essentially two switches with a common pin. 

Here's a really shitty paint scribble of what I mean:

https://puu.sh/xi9UA/ea6fe6a8f2.png

EquilibriumDev, 2017-08-24:
I am interested :)  I think the idea of just using an empty row/column
placement in the matrix is a great idea.

Your schematic helps, although your diodes should probably both point to
the columns, right?
I understand the idea ; my initial confusion was that the overwhelmingly
preferred solution to read encoders is to have a common ground and use pull
ups, while you're proposing to "up" the common ground and read pins A and B
like normal pins. I'll do a few tests tomorrow. If they can be treated as 2
regular switches in the matrix, it'll make this easier.
If that is the case, then the idea of adding them in the keymap with custom
keycodes could add some value.

Thank you both, I have enough info to perform some tests before I try and
generalize this.

Wilba6582, 2017-08-24:
My first pic was wrong, edited it. Diodes are correct now. Note that they're "pointing" in the direction of current flow. The I/O "read" pin is on the column, with an internal pull-up resistor.  The row gets selected, essentially it's "grounding" the encoder's common pin. This will "pull down" a column pin (if there is contact in the encoder).

The pic was supposed to explain that if you use diodes in a switch matrix like this example, the diode goes in the direction of current, i.e. column to row, and it doesn't matter if it is before or after the switch.

EquilibriumDev, 2017-08-24:
I understand the source of my confusion yesterday, basically incomplete knowledge... 
I had assumed that "select row, read column" meant pulling up the row, and reading whether the column was high (so with diodes going along the column). But I forgot that there are no pull-down resistors, so the only way to reliably "read" is to use pull-ups, and ground the selected row. I will need to get familiar with this kind of core QMK logic. But with this, I see that integrating an encoder in the matrix is quite natural.

Your wiring suggestion makes a lot more sense to me now, and should work well, I'll try out some stuff. Even if we can integrate encoders within the matrix, we need to support using other pins, I'll look into both solutions.

Wilba6582, 2017-08-24:
Cool, that was sorta the point of chiming in... encoder handling (keeping track of the quadrature states) can be generalized out, so it can be used either way. It's stuff I was going to do eventually... I think a 60%-ish keyboard with a row of encoders up the top would be pretty cool for music production, esp. if they were routed out as MIDI controllers.

EquilibriumDev, 2017-08-25:
Ok I think I will proceed this way:

 - Allow for 1 encoder by using ENCODER_PIN_A and ENCODER_PIN_B (we could try and immediately support multiple encoders by having a list of pairs or something, but we don't want to take too much time out of the matrix scan)
 - Generalize the logic to N encoders by adding custom keycodes such as CW(x) and CCW(x) to use directly in the keymap (with x being the encoder's number)

Both of these will require a callback, to the style of tap dance with the same kind of index.

d-vogel, 2018-04-24:
Hello there,
I have a very simple implementation integrating the rotary encoder in the regular matrix by wiring it in a unusual way. [Let me know what you think.](https://github.com/d-vogel/qmk_firmware/tree/master/keyboards/simpleDial)

drashna, 2019-03-29:
Rotary Encoder support has officially been added to QMK Firmware.
https://docs.qmk.fm/#/feature_encoders

I'm closing the issue because of that. 
If you feel I'm wrong, or that this should be re-opened, then please do so.

closed 2019-03-29 by drashna