qmk/qmk_firmware #1639: How to compile teensy_one_key with QMK

4sStylZ, 2017-08-25:
Hello,

I try too understand how to compile the QMK firmware for my Teensy 3.2 arm architecture.

My first step is compile the  ChibiOS example keyboard « Teensy lc one key » and follow the [instruction.md](https://github.com/qmk/qmk_firmware/blob/master/keyboards/chibios_test/teensy_lc_onekey/instructions.md)

```
aschmitt@bluebook:~/Projects/qmk_firmware/keyboards/chibios_test/teensy_lc_onekey(master)$ make -f Makefile.3.2
Makefile.3.2:76: ../../tmk_core/tool/chibios/common.mk: No such file or directory
Makefile.3.2:77: ../../tmk_core/tool/chibios/chibios.mk: No such file or directory
make: *** No rule to make target '../../tmk_core/tool/chibios/chibios.mk'.  Stop.
```

My environnement is working with ChibiOS but for TMK. With TMK I can compile the teensy_lc_onekey keyboard and instructions seems to be the same. (Clone the two projects ChibiOS and install gcc-arm toolchain. 

So I don’t understand why I have missed with QMK ?
Is there a better keyboard with a Teensy 3.2 keyboard that I can try ?

4sStylZ, 2017-08-27:
I have launched « Make » for all the qmk keyboards and the Whitefox keyboard wich seems to use ChibiOS is the only one who don’t compile.

4sStylZ, 2017-08-27:
```Making deltasplit75/v2 with keymap protosplit [OK]
Making whitefox with keymap default [ERRORS]
tmk_core/chibios.mk:38: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x7.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:40: lib/chibios/os/hal/hal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:46: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:56: lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1/board.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:57: lib/chibios/os/hal/osal/rt/osal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:59: lib/chibios/os/rt/rt.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:65: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:67: lib/chibios/os/hal/lib/streams/streams.mk: Aucun fichier ou dossier de ce type
quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: Aucun fichier ou dossier de ce type
make[1]: *** Aucune règle pour fabriquer la cible « lib/ugfx/gfx.mk ». Arrêt.
Making whitefox with keymap jetpacktuxedo [ERRORS]
tmk_core/chibios.mk:38: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x7.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:40: lib/chibios/os/hal/hal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:46: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:56: lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1/board.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:57: lib/chibios/os/hal/osal/rt/osal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:59: lib/chibios/os/rt/rt.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:65: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:67: lib/chibios/os/hal/lib/streams/streams.mk: Aucun fichier ou dossier de ce type
quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: Aucun fichier ou dossier de ce type
make[1]: *** Aucune règle pour fabriquer la cible « lib/ugfx/gfx.mk ». Arrêt.
Making whitefox with keymap matt3o [ERRORS]
tmk_core/chibios.mk:38: lib/chibios-contrib/os/common/startup/ARMCMx/compilers/GCC/mk/startup_k20x7.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:40: lib/chibios/os/hal/hal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:46: lib/chibios-contrib/os/hal/ports/KINETIS/K20x/platform.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:56: lib/chibios-contrib/os/hal/boards/PJRC_TEENSY_3_1/board.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:57: lib/chibios/os/hal/osal/rt/osal.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:59: lib/chibios/os/rt/rt.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:65: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/mk/port_v7m.mk: Aucun fichier ou dossier de ce type
tmk_core/chibios.mk:67: lib/chibios/os/hal/lib/streams/streams.mk: Aucun fichier ou dossier de ce type
quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: Aucun fichier ou dossier de ce type
make[1]: *** Aucune règle pour fabriquer la cible « lib/ugfx/gfx.mk ». Arrêt.
Making kmac with keymap default [OK]
Making kmac with keymap winkeyless [OK]
Making orthodox/rev1 with keymap default [OK]
```

4sStylZ, 2017-08-30:
My problem was caused by the two chibiOS directory wich are cloned by git submodule in the `/lib` folder.
That’s not a very bad way for cloning the two dependancies for arm / teensy architecture but I was following another keyboard documentation : 

https://github.com/qmk/qmk_firmware/blob/master/keyboards/chibios_test/teensy_lc_onekey/instructions.md

This documentation seem’s obsolete because the instructions asks to cloning the `ChibiOS` and `ChibiOS-Contrib` in two folders : 

* `tmk_core/tool/chibios/chibios` wich is now synced by the git submodule command in `lib/chibios/chibios`.
* `tmk_core/tool/chibios/chibios-contrib` wich is now synced by the git submodule command in `lib/chibios/chibios-contrib`

I think this keyboard instruction is important because when you try to make a Teensy 3.2 based keyboard you’r searching for the most simple example and the « teensy_lc_onekey » is the most simple.

I will update-it but I have a bad English.

closed 2017-08-30 by 4sStylZ

olivierpons, 2017-09-18:
I have an infinity ergodox, and I have this problem when I try to compile:

make ergodox_infinity-french_hacker
Making ergodox_infinity with keymap french_hacker

quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: No such file or directory
make[1]: *** No rule to make target 'lib/ugfx/gfx.mk'.  Stop.
Makefile:515: recipe for target 'ergodox_infinity-french_hacker' failed
make: *** [ergodox_infinity-french_hacker] Error 1
Make finished with errors

4sStylZ, 2017-09-23:
I think you have a complete different issue : 
- No arm architecture.
- Different keyboard.
- No use of ChibiOS.

You need to create another issue and describe your problem.