qmk/qmk_firmware #1642: audio volume keycodes

surlyjake, 2017-08-25:
A section of the keycode documentation here: https://docs.qmk.fm/keycodes.html mentions what _appears_ to be duplicate info for audio control, however:

Long Name | Short Name | Description
-- | -- | --
KC__MUTE |   | Mute
KC__VOLUP |   | Volume Up
KC__VOLDOWN |   | Volume Down

dont seem to do anything on my systems. But it does seems like these keycodes work properly:

Long Name | Short Name | Description
-- | -- | --
KC_AUDIO_MUTE | KC_MUTE |  
KC_AUDIO_VOL_UP | KC_VOLU |  
KC_AUDIO_VOL_DOWN | KC_VOLD |  

Is this just old info that needs to be removed or just some clarification needed about what each does?

In ether case, I can submit a pull request to fix/clarify.

barrar, 2017-09-09:
KC__MUTE, KC__VOLUP and KC__VOLDOWN are for mac only. Make a pull request to fix the description. Specify that KC_VOLU works on windows and mac.

surlyjake, 2017-09-22:
Thanks for the clarification.
#1750 Submitted.

closed 2017-09-22 by surlyjake