qmk/qmk_firmware #1650: Let's Split v2: EE_HANDS does not seem to work

davidstosik, 2017-08-28:
I just finished assembling my Let's Split using Wootpatoot's v2 PCB, and two cheap Arduino Pro Micro clones I got on AliExpress.

Overall, the keyboard seems to work very well when set in `MASTER_LEFT` mode and plugging the left half. I wanted to try the `EE_HANDS` setting to get the keyboard to work with either half plugged in (and maybe more later on). Here's what I did (I run on Mac):

1. `cd keyboards/lets_split/keymaps/default`
2. Comment out all `#define MASTER_` lines.
3. Uncomment the line `#define EE_HANDS` in `config.h`
4. Use `make avrdude` to flash both halves with the new firmware (may that have any importance, I unplugged the TRRS cord while flashing).
5. Plug-in the left half, shunt the Arduino's `RST` pin, then flash the EEPROM with `avrdude -p atmega32u4 -P /dev/tty.usbmodem144131 -c avr109 -U eeprom:w:../../eeprom-lefthand.eep` (so the file that can be found in the `lets_split` folder).
6. Repeat with right half, changing the file to `eeprom-righthand.eep`.

Each flash operation seems to complete successfully (no visible error message). I also noticed that after flashing the left side's EEPROM, RGB underglow turns solid red, and when I flashed the right side's EEPROM, a couple LEDs turned on and yellow.

When I plug everything back together:

- keyboard work correctly, as before, when USB is on the left side
- when USB is plugged on the right side, keyboard is completely reversed (QWERTY from right to left, starting from right-side half)

Oh, please also note that I'm using `SERIAL` and not `I2C`, using the extra TRRS pin and the Arduino's TX0 pin for the RGB underglow.

Can anyone please help me figure out what's wrong?

davidstosik, 2017-08-28:
A bit more info:

```bash
# Write file to Arduino's EEPROM
$ avrdude -p atmega32u4 -P /dev/tty.usbmodem144111 -c avr109 -U eeprom:w:../../eeprom-lefthand.eep

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: reading input file "../../eeprom-lefthand.eep"
avrdude: input file ../../eeprom-lefthand.eep auto detected as Intel Hex
avrdude: writing eeprom (11 bytes):

Writing | ################################################## | 100% 0.04s

avrdude: 11 bytes of eeprom written
avrdude: verifying eeprom memory against ../../eeprom-lefthand.eep:
avrdude: load data eeprom data from input file ../../eeprom-lefthand.eep:
avrdude: input file ../../eeprom-lefthand.eep auto detected as Intel Hex
avrdude: input file ../../eeprom-lefthand.eep contains 11 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 11 bytes of eeprom verified

avrdude: safemode: Fuses OK (E:CB, H:D8, L:FF)

avrdude done.  Thank you.

# Compare Arduino's EEPROM to file (I need to reset the Arduino)
$ avrdude -p atmega32u4 -P /dev/tty.usbmodem144111 -c avr109 -U eeprom:v:../../eeprom-lefthand.eep

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: verifying eeprom memory against ../../eeprom-lefthand.eep:
avrdude: load data eeprom data from input file ../../eeprom-lefthand.eep:
avrdude: input file ../../eeprom-lefthand.eep auto detected as Intel Hex
avrdude: input file ../../eeprom-lefthand.eep contains 11 bytes
avrdude: reading on-chip eeprom data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xed != 0x00
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:CB, H:D8, L:FF)

avrdude done.  Thank you.
```

davidstosik, 2017-08-28:
Looks like it works if I disable `RGBLIGHT_ENABLE` in my `Makefile`. Is that a big?

davidstosik, 2017-08-28:
I ended up doing this:

- flash `eeprom-lefthand.eep` on left side's EEPROM
- flash `eeprom-righthand.eep` on right side's EEPROM
- flash your firmware (with `RGBLIGHT_ENABLE=yes` as required) on the left side
- flash whatever firmware you want on the right side (can be the same as left), but make sure to disable `RGBLIGHT_ENABLE` (it wouldn't work anyway if you're outputting RGB LEDs' data from left Arduino's TX0)

r2d2rogers, 2017-08-28:
I think #1576 would apply here, has that fix been added to the Let's Split?

davidstosik, 2017-08-29:
Good catch!
I hope very much that #1600 leads to some factorization, but in the meantime, I think this should fix it for Let's Split: #1655 .

davidstosik, 2017-08-31:
Fixed in #1655

closed 2017-08-31 by davidstosik