qmk/qmk_firmware #1659: [Discussion] Simplify our build infrastructure by eliminating Makefiles

skullydazed, 2017-08-30:
Currently the way our Makefiles are included leads to a lot of complexity. Here are a short list of the problems we have encountered:

* New users are often confused as to whether they should use `rules.mk` or `Makefile`
* The keymap Makefile is parsed twice when make is run at the keymap level, leading to hard to track down bugs.
* Keymaps only support `Makefile`, leading to confusion when a user tries to use `rules.mk` at that level
* It can be confusing for some users to figure out why a make command that works in one directory doesn't work in another
* The logic for determining keyboard/revision/keymap based on $PWD is fragile

To address these issues we'd like to eliminate `Makefile` at the keyboard/keymap level. This will mean that users always run make from the top level `qmk_firmware` directory, and that the full `make keyboard-revision-keymap` command be used. We've prepared the [single_makefile](https://github.com/qmk/qmk_firmware/tree/single_makefile) branch to demonstrate how this will be implemented and give people a chance to try it out first.

Given the wide-ranging impact of this change we'd like to solicit feedback from users. This issue will be open until at least Sep 7 to allow for discussion.

NoahAndrews, 2017-08-31:
Sounds good to me

henxing, 2017-08-31:
I like the idea. I'll try to play with it to see if I like the implementation. Thanks for the transparency!

nooges, 2017-08-31:
I'm cool with this, I normally run the full make command anyway these days, while having the whole project loaded up in Vim or ST.

chief-wrench, 2017-08-31:
This makes it easier to understand for noobs. Have my +1

mfs, 2017-08-31:
I like the idea. I find building from the top directory simple.

jotix, 2017-08-31:
totally agree +1

cbbrowne, 2017-08-31:
I gave the branch a try, and it certainly does seem to be working OK.

One issue with this is that the revision doesn't seem nearly obvious enough.  I had to poke around more than left me happy to figure out I needed "make planck-rev4-cbbrowne".  "planck" and "cbbrowne" were no problem; should it be rev3 or rev4????

jackhumbert, 2017-08-31:
@cbbrowne it sounds like that's more my fault ;) rev3 is the pre-speaker version of the PCB, and has audio disabled. rev4 has it enabled - that's the only difference! It won't hurt a rev3 board to put rev4 firmware on it though.

I thought we had some way to leave out the revision in the make system, and use the default one specified by the keyboard - I'm not sure how these changes affect that.

cbbrowne, 2017-08-31:
@jackhumbert I'm not at all concerned about there *being* a rev3 and rev4; it's just that we need enough clues to be able to readily figure out that we need to specify one of those values.

When I pulled the branch, I suddenly realized, "hey, what am I supposed to use for a revision value?"  Having a clear place to get that is the crucial thing.  It probably suffices to say, "oh, there's a directory keyboards/planck/rev3 and keyboard/planck/rev4; you need to pick one of those."

There deserves to be a readme.md file in those directories so we have some documentation to help distinguish why one would prefer one or the other; that's where audio would get mentioned.

Maartenwut, 2017-08-31:
I like this as well, less clutter 👍

Xyverz, 2017-08-31:
I had to read this twice in order to understand what you're doing here. I'll give this branch a try from home over the weekend and see what happens.

Instead of the per-keyboard/keymap makefile, where are we to put our tweaks? (disable audio, enable rgb, etc...)

skullydazed, 2017-09-02:
@Xyverz How can I make that easier to parse?

Your changes should go into rules.mk once this change has been applied. For everything currently in the repo that will be done for you.

Xyverz, 2017-09-02:
That seems clear enough for me, thank you.

dungdung, 2017-09-09:
Now that we have to type out `make keyboard-subproject-keymap-target` it would be nice to have tab completion. Is it possible to add that to QMK or is it something users would need to set up on their own depending on their environment?

I currently get the following completion entries after typing `make `
```
$ make 
all                     git-submodule           message.mk              test-clean
all-keyboards           make                    quantum/                testlist.mk
all-keyboards-defaults  Makefile                test                    tmk_core
```

skullydazed, 2017-09-09:
That's something that you'd have to setup for your environment, this document explains how to do that: http://www.tldp.org/LDP/abs/html/tabexpansion.html

If you or someone else wanted to write a bash script that could be included in someone's `.profile` we'd love to include it in the `util` directory.

jackhumbert, 2017-09-09:
I'd love to have something like that :) we could even have it run on a make
rule right? Since we're already using a full list of keyboards/keymaps
somewhere.

jackhumbert, 2017-09-10:
@dungdung I started this in #1700 - it supports <keyboard>-<subproject>-<keymap> :)

closed 2017-10-22 by skullydazed