qmk/qmk_firmware #1661: Reading fuses to determine bootloader start

jackhumbert, 2017-08-30:
This information doesn't really need to be an option in QMK - it's configurable via the fuses, and if we read from that, we'd be doing the same thing that the actual reset button does (it's stored in the HIGH byte):

```c
#include <avr/boot.h>;

uint8_t high = boot_lock_fuse_bits_get(GET_HIGH_FUSE_BITS);
if (high & 0b110)
// 256
else if (high & 0b010)
// 512
else if (high & 0b100)
// 1024
else
// 2048
```

drashna, 2018-03-25:
I think that you fixed this already? 
https://github.com/qmk/qmk_firmware/blob/7c9d5ace143d3cc6d767a354acde814926d566fd/tmk_core/common/avr/bootloader.c#L81-L93

drashna, 2018-10-21:
@jackhumbert Okay if I close this?

closed 2019-11-21 by drashna