qmk/qmk_firmware #1662: Build issues when trying to update keymap on Ergodox Infinity

packman86, 2017-08-31:
I was having some issues with one of my keys and decided to build a new keymap.
After downloading the latest version of QMK, I cannot make my keymap anymore.

Can someone help me resolve this issue?

This is my input:

make ergodox_infinity-doxxy

This is my output:

Making ergodox_infinity with keymap doxxy

quantum/visualizer/visualizer.mk:77: lib/ugfx/gfx.mk: No such file or directory
make[1]: *** No rule to make target 'lib/ugfx/gfx.mk'.  Stop.
Make finished with errors
make: *** [Makefile:525: ergodox_infinity-doxxy] Error 1

olivierpons, 2017-09-18:
I have the same problem.

fredizzimo, 2017-09-19:
It sounds like you haven't updated the git submodules properly.

You can either run `make git-submodule`

or the following two commands
```c
git submodule sync --recursive
git submodule update --init --recursive
```

If that doesn't help, I need the output of
`git status`

Benjamin-Blackswan, 2018-01-13:
W530-i7@W530 MINGW64 /c/users/w530-i7/downloads/qmk_firmware/qmk_firmware
$ make git-submodule
': not a valid identifierline 88: export: `dashless
fatal: Not a git repository (or any of the parent directories): .git
make: *** [Makefile:579: git-submodule] Error 128

W530-i7@W530 MINGW64 /c/users/w530-i7/downloads/qmk_firmware/qmk_firmware
$ git status
fatal: Not a git repository (or any of the parent directories): .git

drashna, 2018-03-25:
Did you download the zip, or use `git clone`?

drashna, 2018-10-21:
Abandoned, or fixed.

If not, please reopen.

closed 2018-10-21 by drashna

Benjamin-Blackswan, 2018-10-21:
Abandon Ergodox altogether. Piece of junk and waste of money

olivierpons, 2018-12-16:
The command that worked for me (as time of writing) is `git submodule update --init --recursive`

nmall, 2019-01-24:
I'm willing to bet @Benjamin-Blackswan probably downloaded the zip from the configurator rather than cloning the repo.

I have run the commands and was able to compile an ergodox_infinity fw.