qmk/qmk_firmware #1664: Whitefox Failure

adamclerk, 2017-09-01:
Just pulled down the repo and I'm trying to deploy it to my whitefox

```
dfu-util -D whitefox_matt3o.hex -S mk20dx256vlh7            74 ↵
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%       109235 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
```

After it is flashed it sits there without the LEDs on and doesn't work

```
dfu-util --list
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Found Runtime: [05ac:8289] ver=0137, devnum=6, cfg=1, intf=3, path="20-3", alt=0, name="UNKNOWN", serial="UNKNOWN"
Found DFU: [1c11:b007] ver=0000, devnum=7, cfg=1, intf=0, path="20-3", alt=0, name="Kiibohd DFU", serial="mk20dx256vlh7"
```

I then flash a fresh copy of the firmware from the whitefox configurator to get it back up.

Any thoughts or help would be appreciated.

kimkimkeren, 2017-09-03:
Well, according to thread #987 , you have to compile and flash like this : 
```
make whitefox-default-dfu-util
dfu-util -D ../../.build/whitefox_default.bin  -S mk20dx256vlh7
```

Example : (sudo doesn't required)
```
kim-kim@kimkim-Inspiron-7447:~/qmk_firmware/keyboards/whitefox$ sudo make whitefox-default-dfu-util
QMK Firmware v0.5.119
Making whitefox with keymap default and target dfu-util

arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  38744	      0	  38744	   9758	whitefox_default.hex

Compiling: ./tmk_core/common/command.c                                                              [OK]
Linking: .build/whitefox_default.elf                                                                [OK]
Creating binary load file for Flash: .build/whitefox_default.bin                                    [OK]
Creating load file for Flash: .build/whitefox_default.hex                                           [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  38744	      0	  38744	   9758	whitefox_default.hex

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: More than one DFU capable USB device found! Try `--list' and specify the serial number or disconnect all but one device

tmk_core/chibios.mk:157: recipe for target 'dfu-util' failed
make[1]: *** [dfu-util] Error 74
Make finished with errors
../../Makefile:513: recipe for target 'whitefox-default-dfu-util' failed
make: *** [whitefox-default-dfu-util] Error 1
kim-kim@kimkim-Inspiron-7447:~/qmk_firmware/keyboards/whitefox$ sudo dfu-util -D ../../.build/whitefox_default.bin  -S mk20dx256vlh7
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1c11:b007
Run-time device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 1024
Copying data from PC to DFU device
Download	[=========================] 100%        38748 bytes
Download done.
state(7) = dfuMANIFEST, status(0) = No error condition is present
dfu-util: unable to read DFU status after completion
kim-kim@kimkim-Inspiron-7447:~/qmk_firmware/keyboards/whitefox$
```

adamclerk, 2017-09-05:
That was it. I used the .bin file in the .build folder. Thing worked as expected. Thanks.

closed 2017-09-05 by adamclerk