qmk/qmk_firmware #1668: What's the _best_ way to share code between keymaps

jasonm23, 2017-09-02:
I'm attempting to DRY up the code for my various keyboards, however I'm using a method that would make most C programmers cry.

In a nutshell I simply extracted all the shared code out into a `.inc` file and use `#include "shared.inc"` in my keymaps.

```
qmk_firmware/
   |- shared.inc
   |- ... (everything else)
   `- keyboards 
        `- frosty_flake
             `- keymaps
                  `- mine / keymap.c <- each keymap has  a #include "shared.inc"
        `- gonnerd
             `- keymaps
                  `- mine / keymap.c <- #include "shared.inc"
        `- phantom
             `- keymaps
                  `- mine / keymap.c <- #include "shared.inc"
        `- etc..
             `- keymaps
                  `- mine / keymap.c <- #include "shared.inc"
```

When I tried to do this in a nice way, e.g. create a `shared.c` and `shared.h`

```
qmk_firmware/
   |- shared.inc
   |- ... (everything else)
   `- keyboards 
        `- frosty_flake
             `- keymaps
                  `- mine / keymap.c <- each keymap has  a #include "shared.h"
        `- gonnerd
             `- keymaps
                  `- mine / keymap.c <- #include "shared.h"
        `- phantom
             `- keymaps
                  `- mine / keymap.c <- #include "shared.h"
        `- etc..
             `- keymaps
                  `- mine / keymap.c <- #include "shared.h"
```
and declare functions which handle actions and macros, the keyboard make command compiles, but the macros and actions don't run.

For the moment I am fine using the `#include "shared.inc"`, but it seem that there must be a better way to do this! (that won't cause God to kill any more kittens!)

Wilba6582, 2017-09-04:
Are you adding `shared.c` to the `rules.mk` file? i.e. adding it to `SRC = <files>`

Just guessing here... if it was missing, you'd expect link errors if they are functions being used but their definitions not being compiled/linked...

jasonm23, 2017-09-04:
If that's the way it's done officially, thank you.

I was only doing what I'd described in the issue description.

Wilba6582, 2017-09-04:
The make process will automatically compile the `<keyboardname>.c` file, and the `keymap.c` file. What you want is to add extra .c files to the build.

Here's an example: https://github.com/qmk/qmk_firmware/blob/master/keyboards/lets_split/rules.mk

You should copy the `SRC += ...` part to suit your own requirements. 

Not sure what you mean by this method not being the usual idiomatic C way. Makefiles are supposed to have a list of all the .c files that should be compiled. In QMK, there's a parent/generalized Makefile which will automatically include .c files of certain names; if you want more, you have to add them yourself to the list of .c files to compile.

jackhumbert, 2017-09-04:
#1559 tries to help with this by creating /users/<user> folders where folks can keep common code for multiple keyboards - I'm not sure there's a consensus on the implementation, though.

jasonm23, 2017-09-05:
Hi @Wilba6582

> Not sure what you mean by this method not being the usual idiomatic C way. Makefiles are supposed to have a list of all the .c files that should be compiled. 

My C experience is rusty, prior to messing around with QMK.  The last time I wrote C was in the early part of the 2000s.

I had constructed Makefile as a place to link .o files (I guess I forgot that make infers that as .c when the .o is missing!)

Anyway many thanks for your help!  I've forgotten an awful lot about C so I'll have an annoying habit (to myself) of asking silly questions and then suddenly remembering some obscure details about the exact same thing.

@jackhumbert that is useful to know, I guess it will take people providing sample code for the `/users/`folder.

I will be happy to put some shared macros / action functions in there as an example.

jasonm23, 2017-09-07:
Hi @wilba6582 unfortunately using make to provide src of shared.c doesn't work.

It seems that a keymap.c must provide action functions directly (not sure why) so only an #include which directly adds in the shared functions does that ...

Wilba6582, 2017-09-07:
You need to show us the source, point us at a fork or something. AFAIK there's no technical reason why you can't structure things as a shared .h and .c files.

jasonm23, 2017-09-07:
When  I add the src I forgot to put the include in the keymap.

I'll have a look at it later, travelling without a programmable keyboard for a few days

drashna, 2017-12-05:
If you're still having issues with this, you should check out the (recently) created example:
https://docs.qmk.fm/feature_userspace.html#example

Or you can take a look at my ... fairly large userspace files:
https://github.com/qmk/qmk_firmware/tree/master/users/drashna

closed 2018-03-25 by drashna