qmk/qmk_firmware #1671: Documentation Translations

jackhumbert, 2017-09-03:
We've had some interest in translating the QMK documentation into different languages, so I'd like to see how much interest there is to help with this, or ideas/suggestions on the format.

Gitbook [has support for multiple languages](https://toolchain.gitbook.com/languages.html), so if we move the current English docs to `/docs/en/`, we can establish folders for each translation. (I previously mentioned that we would keep the same English urls/filenames, but I'm not sure this is a standard/makes sense)

My current thinking is that we could use a translator service to jump-start the structure, and let community members make corrections from there - the main question is how many/which languages we would want to support.

If you're interested in helping with this, please let us know along with the languages you'd be comfortable with translating!

Dbroqua, 2017-09-04:
Hello,

This is a great idea!

However I think using a translator service (google translate?) is not a good idea because the result will be too approximative.

I can help you for the translation into French.

jackhumbert, 2017-09-04:
Great! Yeah, using a translator service wasn't meant to be perfect, but I think it can be a good start for pages that the people (actual translators) haven't had the chance to look at yet.

Dbroqua, 2017-09-04:
The translator service can create mistranslations or mistakes. I think that for people who want to translate again after a first translation done by Google trad it will be harder than if they translate by themselves. If they translate by themselves they wouldn't have to check each meaning and to look for potential mistakes, mistranslations and so on.

hiro2016, 2017-09-08:
I don't know If an outsider like myself should barge in here but....   

 - code savvy people generally understand English and they can just read the source  
 - casual users would probably like GUI config tool over technical docs that do not make much sense  
  
I'm somewhere between casual and code savvy but I want better English docs over docs translated into my native language as translated docs in general make less sense.  
  
As to configuration tool I cobbled up one for ergodox:https://github.com/hiro2016/ergodox_gui_configurator      
I needed the above because online eargodox config tools are for english-us only; the tool is incomplete and has no macro support as of now and ugly.
I don't feel like writing a generic qmk config tool but there should be no technical challenges;just a lot of boring work.

namelivia, 2019-10-04:
I am looking for some hacktoberfest contributions and I can translate from English to Spanish. So if that's ok for you I'll start working on it.

yanfali, 2019-10-04:
@namelivia if you have some time please help review https://github.com/qmk/qmk_configurator/pull/516

4sStylZ, 2019-10-04:
I am actually doing the same for the french. I will make a PR when it will be tested.

Actually I have all the newb doc and summary translated.

namelivia, 2019-10-05:
I started also with a fork and a branch (if anyone interested and want to help [here is the link](https://github.com/namelivia/qmk_firmware/tree/spanish_translation/docs/es)).
Should I also open an issue for tracking progress?

noroadsleft, 2019-10-05:
@namelivia, @kbravh has also started working on Spanish docs translations. Perhaps you two can team up?

I've pinned #6924 for tracking which translations (both project and language) that people are working on.

kbravh, 2019-10-05:
@namelivia I'll start working on your fork so we can consolidate the Spanish translations there!

namelivia, 2019-10-15:
The newbie guide is almost ready, both written and reviewed by @kbravh and me (https://github.com/namelivia/qmk_firmware/pull/2/files) Would you think it would be convenient opening a PR with that? We can continue after

noroadsleft, 2019-10-18:
Should be able to just open a pull request from https://github.com/namelivia/qmk_firmware/tree/spanish_translation to https://github.com/qmk/qmk_firmware/tree/master.

paryz1i12, 2019-10-19:
I'd love to help with translation to Polish. @jackhumbert If you're interested in that I can start working on it right away :)

closed 2020-02-16 by stale[bot]