qmk/qmk_firmware #1673: Does Knight Rider mode need a fix?

davidstosik, 2017-09-03:
Hello, I'm currently confused by the *Knight Rider* RGB animation, as it does not seem to do what I would expect it to by default.

First of all, I'm assuming the mode's full name is *Knight Rider* after seeing naming in the code (particularly [`rgblight_effect_knight()`](https://github.com/qmk/qmk_firmware/blob/f0db2c0512996c30372a586aa535a6dca66dce0a/quantum/rgblight.c#L541)).

Secondly, in case some people don't know yet, here is Knight Rider:
[image]

And KITT's red scanner lights:
[image]

Now let's have a look to what I see when I set my Let's Split on *Knight* mode, without any related change to QMK's code:

[image]

I dug a little bit in the code, and found a few configuration variables.

https://github.com/qmk/qmk_firmware/blob/f0db2c0512996c30372a586aa535a6dca66dce0a/quantum/rgblight.h#L29-L38 

> I wonder what that offset can be...? 🤔

https://github.com/qmk/qmk_firmware/blob/f0db2c0512996c30372a586aa535a6dca66dce0a/quantum/rgblight.c#L569-L576

Looks like it's offsetting (duh!) the calculated output before refreshing the LEDs. Note that if `RGBLIGHT_EFFECT_KNIGHT_OFFSET` is `false` or `0`, then `led` never gets new values assigned, and the LED output never changes. That, to me, is the first weird thing. Shouldn't it be something like this?

```c
  for (i = 0; i < RGBLED_NUM; i++) {
    cur = (i + RGBLIGHT_EFFECT_KNIGHT_OFFSET | 0) % RGBLED_NUM;
    led[i].r = preled[cur].r;
    led[i].g = preled[cur].g;
    led[i].b = preled[cur].b;
  }
```

I'd even say that `RGBLIGHT_EFFECT_KNIGHT_OFFSET` should always be a number and remove the weird ` | 0`.

Note how, if `RGBLIGHT_EFFECT_KNIGHT_OFFSET` is equal to `0`, no offset is applied, and yet `led` actually gets changed!

Before changing any C code, I tried playing with that value a little bit. I tried setting it to the number of LEDs I have:

```c
#define RGBLIGHT_EFFECT_KNIGHT_OFFSET RGBLED_NUM
```

Thanks to the modulo `%` in the code above, this should be identical to no offset at all.

[image]
*I was right! (We're still a bit off and do not match KITT's scanner exactly because of too long a pause at the ends, but that'll do.)*

Alright, so, it looks like I identified what the offset value does (offset the starting and ending points of the back and forth movement), and how it works.

I'd like to propose a double fix:

- Fix the C code as I suggested above, to make it a bit more succinct and obvious (also removing the bug when offset is set to `0`).
- Change the offset's default value to 0. Unless someone can point me to a real-life use case that requires a different value, and that's common enough to justify it being the default, no offset is what makes the most sense to me.

... What's that? Another macro name caught your attention? `RGBLIGHT_EFFECT_DUALKNIGHT_LENGTH` looks like it could bring a cool mode to split keyboards (and non split ones too I guess), but unfortunately, this macro is used nowhere in QMK's code... :(
I'd suggest removing it until someone actually implements a new mode.

Sorry for the long post, hope everything is clear and makes sense!

davidstosik, 2017-09-03:
As an addendum to the fix above, I'd also like to suggest an improvement to the *Knight Rider* mode. As I hinted above, the pause at each end feels too long. This could be a remedy:

```diff
diff --git i/quantum/rgblight.c w/quantum/rgblight.c
index d568a4f4..9133c459 100644
--- i/quantum/rgblight.c
+++ w/quantum/rgblight.c
@@ -575,14 +575,14 @@ void rgblight_effect_knight(uint8_t interval) {
   rgblight_set();
   if (increment == 1) {
     if (pos - 1 < 0 - RGBLIGHT_EFFECT_KNIGHT_LENGTH) {
-      pos = 0 - RGBLIGHT_EFFECT_KNIGHT_LENGTH;
+      pos = 0;
       increment = -1;
     } else {
       pos -= 1;
     }
   } else {
     if (pos + 1 > RGBLED_NUM + RGBLIGHT_EFFECT_KNIGHT_LENGTH) {
-      pos = RGBLED_NUM + RGBLIGHT_EFFECT_KNIGHT_LENGTH - 1;
+      pos = RGBLED_NUM - 1;
       increment = 1;
     } else {
       pos += 1;
```

[image]

skullydazed, 2017-09-03:
Put together another PR and we can merge it. :)

davidstosik, 2017-09-03:
Sorry for the double post. I'll close #1672.

I'll make a PR to include the code above.

skullydazed, 2017-09-06:
Thanks for your work on this!

closed 2017-09-06 by skullydazed