qmk/qmk_firmware #1678: Keypresses When Board Is Plugged In

zachfedor, 2017-09-04:
I have a Let's Split flashed with [this layout](https://github.com/zachfedor/qmk_firmware/tree/master/keyboards/lets_split/keymaps/zach). When I plug my keyboard into my laptop, it registers a few keypresses:

```
down: Ctrl + A
up: Ctrl
up: A
down: Tab    // this 4 key combo repeats 2 more times
down: Q
down: R
down: T
up: Tab
up: Q
up: R
up: T
```

Does anyone know what might cause this? Or is there someway to delay keypresses for a certain duration to avoid the bug?

jackhumbert, 2017-09-14:
Do you have your right side or left side plugged into the computer? I wonder if this is something with the logic that makes the sides talk to each other, since they all come from the left side.

zachfedor, 2017-09-14:
I have the right side plugged in. There was some issues with the left side as master (in more detail [here](https://github.com/nicinabox/lets-split-guide/issues/12)), so I had to do some overrides and force right hand as master. That's an interesting thought though, I hadn't made that connection.

drashna, 2018-03-25:
The Ctrl A is odd, but the rest looks like it's in the same row.   Could be the logic, but could be an issue with the row (a short somewhere, or a cold joint maybe).

Either way, were you able to resolve this?

closed 2018-10-21 by drashna