qmk/qmk_firmware #1679: MT(MOD_LALT, MOD_RALT) not working

matz-e, 2017-09-04:
Hi,

I'm trying to get the equivalent of TMK's

    ACTION_MODS_TAP_KEY(MOD_LALT, KC_RALT)

which I use to distinguish between the ALT modifier (left alt) and the compose key (right alt), However,

    MT(MOD_LALT, KC_RALT)

only shows me the keycode for the first key under `xev`. In this case left alt, both on press and tap. If I switch the keys or substitute other mods, I get the same result, one key for both. Any help would be appreciated.

Keymap [here](https://github.com/matz-e/qmk_firmware/blob/4e6962b8dd563cabc478b8e1cbfe86f04c319d00/keyboards/handwired/happy_planck/keymaps/default/keymap.c#L38-L39), using a teensy3.2.

drashna, 2018-03-25:
Since I'm american (and dump to the ways of AltGr), wouldn't you need to hold down RALT here? 

Also, MT is an alias/macro for ACTION_MODS_TAP_KEY(). 

Also, there are a number of options that may help here: z
https://docs.qmk.fm/config_options.html
In the `Behaviors That Can Be Configured` section.

matz-e, 2018-03-25:
Thanks for the notes! I'll see if the option yield anything. Expanding a bit on my issue:

For the "conventional usage" of AltGr, I would have to hold it down. Since I'm pretty used to an American "standard" (I shuffle Ctrl, Alt, CapsLock, etc around) layout, I'd prefer typing with it. What I'd like to achieve is basically:

* Have one "Alt" key (think planck-like keyboard with only one modifier of each kind):
   * when pressed acting like left alt
   * when tapped acting like a [compose key](https://en.wikipedia.org/wiki/Compose_key)

Examples of the compose key (it's sort of a linux thing mostly IMO) — may be wrong/off, keys are pressed one after another, not at the same time:
   * Compose, `a`, `:` = `ä`
   * Compose, `.`, `.` = `…`
   * Compose, '=', `!` = `≠`

I have it on my list to look into this a bit more with debug mode. My other tap combination, Shift/Enter works as expected.

fredizzimo, 2018-04-07:
Yes, modifiers don't work as parameters to the `MT` macro. That's a limitation caused by the fact that we use 16-bit keycodes. 

I'm planning to add support for that in another way, so hopfully it will be supported soon.

closed 2018-10-21 by drashna

matz-e, 2018-10-27:
For reference, I'm currently working around this by setting my compose key to PrintScreen and using that for the tap. Not as flexible as right alt (given other OS will see a PrintScreen rather than an Alt keycode), but working.